mne-tools/mne-python #5073: Issues in importing Neuroscan EEG raw data(.cnt)

voidsoundist, 2018-04-01:
Hi all, 
I'm new for MNE.
I've been trying to import a Neuroscan data file (.cnt) in the past few days, but I've had a problem that I can not solve by myself.
(_In fact, when using the **mne.io.read_raw_cnt()** function, it was difficult to find a montage match with the 64channel montage used in our experiment, but this was a temporary fixed(If someone has resolved similar issues related to this, could you give me some tips?)._)

At first, an error occurred because montage was not matched. But now we have solved that, but there are still other problems.

I guess this is not a problem with MNE. Maybe there might be a problem in creating a virtual environment using anaconda. So I created a new virtual environment and started from MNE install again, but the still same problem occurs.
Here are my code and outputs. And I also attach the example EEG raw file(.cnt).
I'm sorry to you because it seems a basic level. However, I've tried quite a few searches and can not resolve it, so I am waiting for your help. Thanks

`raw = mne.io.read_raw_cnt(filename, montage = 'standard_1020', eog = ['HEO', 'VEO'], ecg = ['EKG'], emg = ['EMG'], misc = ['Trigger', 'CB1', 'CB2'])`

> /anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
<stdin>:1: RuntimeWarning:   Could not parse meas date from the header. Setting to None.
  Could not parse meas date from the header. Setting to None.
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
[... 74 lines cut ...]
  data = np.fromstring(string, dtype=dtype)[0]
[... cut ...]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]
/anaconda2/envs/mne/lib/python2.7/site-packages/mne/io/utils.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(string, dtype=dtype)[0]

[exampleCNTfile.zip](https://github.com/mne-tools/mne-python/files/1866539/exampleCNTfile.zip)

agramfort, 2018-04-01:
I see just see warnings due to very recent numpy.

do you actually get an error?

voidsoundist, 2018-04-01:
Thanks, agramfort!

OMG... does it just warning?...
I've just checked it, so... yes, as you said, I can check the data was read well...

output:
`<RawCNT  |  noise_.cnt, n_channels x n_times : 70 x 32900 (32.9 sec), ~245 kB, data not loaded>`

If this kind of thing happens every time, then is it ok?
I mean, do I have to change a version of Numpy? or others?

agramfort, 2018-04-01:
Yes Its Just warnings you can ignore and that will be fixed with new release

closed 2018-04-01 by agramfort

voidsoundist, 2018-04-02:
Thanks a lot, Agramfort!!

SunJingmin, 2021-11-17:
hello, I have the same problem. Have you solved it?

drammock, 2021-11-17:
@SunJingmin this thread is over 3 years old.  I recommend you (1) make sure you're using the current version of MNE-Python (see our [install instructions](https://mne.tools/stable/install/index.html)), and (2) if you still get this error, open a new issue that includes a link to a small file that causes the error.  In the issue description please reference this issue (typing `#5073` in the PR description will create the cross-reference automatically).