mne-tools/mne-python #5088: coreg issue in mne 0.16

dokato, 2018-04-04:
When running with mne-0.16.dev0: `mne coreg` I got: 

```
TypeError: coregistration() got an unexpected keyword argument 'project_eeg'
```

dokato, 2018-04-04:
Strange, I do have different mne versions but in separate conda environments.

christianbrodbeck, 2018-04-04:
Did you check your paths? It looks like you might be accessing the script (the `$ mne` command) from a different version than the package. Also check `$ which mne`.

christianbrodbeck, 2018-04-07:
I think it could be due to mixing conda and pip installs, they're not completely compatible, I would try (pip) installing mne again.

larsoner, 2018-04-09:
Closing as an installation problem, but we can reopen if there is something to fix at the MNE end.

If you do figure it out, maybe adding some insights to a new "debugging conflicting environments" entry in the FAQ would help.

closed 2018-04-09 by larsoner