mne-tools/mne-python #5102: BUG: read_epochs for split files has incorrect events

emilyps14, 2018-04-05:
I have a large epochs file, that gets split into two files when I call epochs.save()
When I load it again using mne.read_epochs(), the events corresponding to the second file are incorrect, offset by [last epoch from the first file + tmax + 10s].

The bug comes from _concatenate_epochs.

This breaks the relationship between the raw file and the events list, which (aside from causing downstream problems with my code) can lead to an error in epochs.get_data if the data were loaded with preload=False.

agramfort, 2018-04-05:
can you share a code snippet to replicate usings sample data? you can make
random events if it's just for testing.

emilyps14, 2018-04-05:
`
import os.path as op
import mne

data_path = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample')
raw = mne.io.read_raw_fif(
    op.join(data_path, 'sample_audvis_raw.fif'), preload=True)

events = mne.make_fixed_length_events(raw,1)
epochs = mne.Epochs(raw, events)
epochs.save('test-epo.fif',split_size='100MB')

epochs2 = mne.read_epochs('test-epo.fif',preload=False)
epochs2.get_data() # throws error
assert(all([e1==e2 for e1,e2 in zip(epochs.events[:,0],epochs2.events[:,0])])) # assertion error
`

agramfort, 2018-04-05:
I don't think @larsoner is good enough to fix this :)

larsoner, 2018-04-05:
I see what you did there :)

Yes I can try hopefully next week

emilyps14, 2018-04-05:
Yep, that fixes it. Thanks!

PR #5103 (2018-04-06): MRG: Fix split saving
Closes #5102.
closed 2018-04-06 by agramfort