mne-tools/mne-python #5105: ENH: add precomputed fwd operator for somato?

mathurinm, 2018-04-06:
I am working on IR schemes for TF-MxNE in #5075. The benefits of this solver are best shown on data where the sources contains a mix of transient and slow activations, eg in somato data. For this reason, in the example I plan to add in my PR, I'd like to work on somato instead of sample. 

Currently the forward solution is not stored and has to be computed, which would make the example long to run and less specific about irTFMxNE.

Would it be possible to add the forward solution to the data in the somato dataset, as it is the case in sample ? If so, what is the way to go?

larsoner, 2018-04-06:
This is possible depending on the increase in file size. For forward it can be large but would be worth it IMO. Same process would apply for #4184 and other "dataset-update" requests:

1. Download dataset
2. Extract
3. Add file
4. Compress
5. Upload to `osf.io` (@agramfort can give you upload permissions, or share the file with me or him and we can do it)
6. Open PR to update hash in `mne/datasets/utils.py`
7. Test that the download works on new PR
8. Update any tutorials / examples as necessary

mathurinm, 2018-04-06:
Thanks for the detailed answer, I'll do it.

mathurinm, 2018-04-13:
The .fif file is 36.9 MB. Is the name `somato-meg-oct-6-fwd.fif` OK ?
I've shared it on dropbox with @larsoner and @agramfort, i'll add the tar.gz file

The new  `MNE-somato-data.tar.gz` is 655 MB (was 589 MB) > am I comso ressing right ?
I used `tar -zcvf` to compress the directory.

larsoner, 2018-04-13:
> The .fif file is 46.9 MB. Is the name somato-meg-oct-6-fwd.fif

Yeah that seems fine

> am I compressing right ?

If you take the unmodified uncompressed directory and recompress it, do you get the same file? If not, try tweaking the compression-level option until you do. Based on the 47MB file increasing the size by 66MB I'm guessing it's not high enough.

mathurinm, 2018-04-13:
The increase seems to be due to Nautilus's file properties giving a wrong value
Command line (`ls -l -hr`)  gives me correct values:
```
-rw-rw-r--  1 mathurin mathurin 589M avril 13 15:26 MNE-somato-data.tar.gz
-rw-rw-r--  1 mathurin mathurin 626M avril 13 14:50 MNE-somato-data_with_fwd.tar.gz  
```
where` MNE-somato-data.tar.gz` has been downloaded and recompressed locally and  `MNE-somato-data_with_fwd.tar.gz`   is the version with the forward file.

larsoner, 2018-04-13:
> The increase seems to be due to Nautilus's file properties giving a wrong value

FYI this is a binary (Mi) vs SI (M) prefix issue (2 ** 20 == 1048576)
```
>>> 626 * (2 ** 20 / 1e6)
656.4085759999999
```