mne-tools/mne-python #5108: Interactive plots with show=False

bloyl, 2018-04-09:
Interactive plots with show=False or with non blocking matlabplot backends can lead to unexpected results.

For instance (here is gist showing this for sample data - https://gist.github.com/bloyl/7f6a7910bf2afce5f198b6233f39416b )
```
evoked.info['bads'] = []
evoked.plot(exclude=[], show=False)
evoked.info['bads'] += [new_bad_chans]
evoked,.plot(exclude=[],show=False)
plt.show()
```
This will make 2 figures. the Butterfly plots will be correct, the first will show no bad channels and the second will show the bads chans in red.

However if you select a time range from either plot you will see a topomap created using on the 'good' channels of evoked. This is unexpected for 2 reasons.
  1) exclude=[] should use all the channels. Maybe indicating bad channels somehow.
  2) in figure 1, the topomap doesn't match the butterfly plot because the data used to make it changed but only the interactive part (the topomap) saw that change.

This is understandably a corner case and not a high priority but I wanted it listed so I don't completely forget about it :)

larsoner, 2018-04-09:
Yes I could see how this would happen. I'm not sure if there is a good solution, though. Should we add something to the `plot_visualize_evoked` or somewhere else about this behavior? This will also affect all plotting modes where you can modify an object (in weird/probably breaking ways), such as `epochs.plot` and `raw.plot`.

bloyl, 2018-04-09:
I don't know enough about python to know if this is possible, but I would
think the ideal case would be to monitor the object and then redraw
derivative plots (butterfly, topos etc) if the object is modified in a
meaningful way. An alternative would be store a copy of the object for
plotting, but that would have huge overhead for epochs or raw.

agramfort, 2018-04-09:
at least

evoked.plot

could  be fixed as it cannot change the evoked instance.

drammock, 2020-11-24:
> I don't know enough about python to know if this is possible, but I would think the ideal case would be to monitor the object and then redraw derivative plots (butterfly, topos etc) if the object is [...]

another option to consider here would be disabling interactivity of a figure when `show=False`.  That way, the revealed plot would be accurate *as of the time it was created*, no (incorrect) derivative plots could be produced from it, and the lack of interactivity would sort of reflect its "static-ness".

@bloyl do you often want to initialize a hidden plot, run some other code, and then later call `plt.show()` and interact with the revealed plot?  Or was this a case of "I did this by accident once, and I'm documenting it because it seems bad"?