mne-tools/mne-python #5109: API: Unify lcmv and dics code

larsoner, 2018-04-09:
Some of the inversion / pinv / `eigh` calls should be unified in the LCMV and beamformer code, see XXX's introduced in `_lcmv.py` and `_dics.py` in #5066.

britta-wstnr, 2018-04-10:
I'd be happy to take a look next week - or should is this a milestone for 0.16, @larsoner ?

larsoner, 2018-04-10:
I don't think it should be a blocker for the release, but if we can get it in that would be graet

britta-wstnr, 2018-04-10:
Okay, I will try to get a list of what would be needed ASAP - then maybe we can see how big of a project this will be and if it's reasonable to get it done fast?

larsoner, 2018-04-10:
Sounds good to me

britta-wstnr, 2018-04-13:
As far as I see it, the following needs to be done:
*  functions like ``_prepare_beamformer_input``, ``_setup_picks``, ``_reg_pinv, eig_inv``, ``_check_proj_match``, ``_pick_channels_spatial_filter`` (and whatever I missed at first glance) could go into a ``beamformer_utils`` file
* the beamformer "heart" (the core of ``make_lcmv`` and ``make_dics``) could go to a ``_compute_beamformer`` function that is shared by DICS and LCMV.

The second point is a bit sensitive, since there are some functionalities that are shared between DISC and LCMV and some that work quite differently.

For a first iteration I would keep everything as it is at the moment and only unify truly shared code, but still making it one file. From there we can see where we want to change the respective methods ``lcmv`` and ``dics``. This would not require and API changes.

I am happy to give it a go - if you are okay with this setup @larsoner I would start working on it!

larsoner, 2018-04-13:
Yes sounds good to me as two separate PRs. I'm theory no tests should need
to change just internal calls.

I would also just put all shared code (including that for the first PR) in
the _compute_beamformer.py file for simplicity.

>

britta-wstnr, 2018-04-13:
Okay then I will start with the first one, which should be quick enough because it's all functions, so it's just changing imports and think about how to do number 2 in the meanwhile :wink:
PR(s) follow!

larsoner, 2018-08-18:
Subsumed by #5365

closed 2018-08-18 by larsoner