mne-tools/mne-python #5115: Proposition: add small utility for managing filenames

wmvanvliet, 2018-04-11:
When writing our example pipeline for processing 20 subjects for the Frontiers Research Topic, an excessive amount of lines and complexity became devoted to constructing filenames:

    epochs_fname = subjects_dir + 'sub%02' % subject + '-' + condition + '-epo.fif'

...and so forth. Do clean up the code, I wrote a small `FileNames` class, which ended up being so very useful it is now an integral part in all my analysis scripts.

Here it is: https://gist.github.com/wmvanvliet/66325f052b579d510a7517c5aaf189cb
The documentation on how to use it is pretty small and given at the top of the gist.

The `FileNames` class provides very basic templating functionality. With this, you provide the templates for all your filenames in a file like `config.py`:

```python
# Filenames for all the files that will be generated during the analysis
fname = FileNames()

# Some directories
fname.add('study_path', study_path)
fname.add('meg_dir', meg_dir)
fname.add('subjects_dir', subjects_dir)
fname.add('subject_dir', '{meg_dir}/{subject}')

# Sensor level files
fname.add('epo', '{subject_dir}/{subject}_highpass-{l_freq}Hz-epo.fif')
fname.add('csd', '{subject_dir}/{subject}-{condition}-csd.h5')

# Source level files
fname.add('src', '{subjects_dir}/{subject}/fsaverage_to_{subject}-{sp}-src.fif')
fname.add('trans', '{study_path}/ds117/{subject}/MEG/{subject}-trans.fif')
fname.add('fwd', '{subject_dir}/fsaverage_to_{subject}-meg-{sp}-fwd.fif')
fname.add('power', '{subject_dir}/{subject}-{condition}-dics-power')

# Connectivity files
fname.add('con', '{subject_dir}/{subject}-{condition}-connectivity.h5')

# Grand-average files
fname.add('fsaverage_src', '{subjects_dir}/fsaverage/fsaverage-{sp}-src.fif')
fname.add('ga_power', '{meg_dir}/{condition}-average-dics')
fname.add('ga_power_hemi', '{meg_dir}/{condition}-average-dics-{hemi}.stc')
fname.add('ga_con', '{meg_dir}/{condition}-average-connectivity.h5')
```

And then in your analysis scripts you go:

```python
import mne
from config import fname

# Load the epochs for all subjects
epochs = [mne.read_epochs(fname.epo(subject=s)) for s in subjects]
```

Advantages:

 - No more `op.join` hell
 - Define filenames in __one place__, so they are consistent throughout your code, guaranteed, and when running your code on another machine, there is only a single place where you have to worry about updating the file paths.

### Would you be interested in adding this to MNE-Python, or shall I just keep it as a separate package?

agramfort, 2018-04-11:
I was think of something like this to expose in python the bids files so we
should "brainstorm" how to best converge here.

wmvanvliet, 2018-04-11:
Do we have another issue open somewhere for it?

Otherwise, let me start with the brainstorm:

The bids files could just be an instance of `FileNames`, constructed like this:

```python
def get_bids_fnames(data_path):
    bids = FileNames()
    bids.add('data_path', data_path)
    bids.add('raw', '{data_path}/MEG/{subject}/{subject}-raw.fif')
    return bids
```

That would result in an `FileNames` object with the following fields:

```
bids.data_path  # A string containing the data_path set by the user
bids.raw(subject)  # Function that returns the raw filename of the given subject
```

wmvanvliet, 2018-04-11:
Does BIDS-MEG specify explicit templates for filenames for raw, tsss, etc?

agramfort, 2018-04-11:
yes it does. You have the subject numbers, the run etc.

jasmainak, 2018-04-11:
@wmvanvliet take a look at: https://mne-tools.github.io/mne-bids/

also [this function](https://github.com/mne-tools/mne-bids/blob/master/mne_bids/utils.py#L32) that @choldgraf wrote

jona-sassenhagen, 2018-04-23:
I have something like

```
sourceloc = "../"
in_template = sourceloc + "unprocessed/{}_{}.vhdr"
out_template = sourceloc + "{2}/{0}-{1}-{2}.fif"
```

And then it's just
```
ftype = "ica"  # or 'raw', 'epo', 'eve', ...
subj = "anonymous"
exp = "oddball"
fname = out_template.format(ftype, exp, subj)
```

Though maybe it's more complicated if you need to work with all the source loc stuff.

wmvanvliet, 2018-04-23:
My class is a very small wrapper around that to turn:

    evoked_template.format(x, y, z)

into:

    fname.evoked(x, y, z)

Just a little readability improvement.

wmvanvliet, 2018-04-23:
It also collects the template into a single namespace so you can just do:

    from config import fname

Instead of having to do:

    from config import evoked_template, grand_average_template

etc.

larsoner, 2021-02-19:
I think this is `mne-study-template` territory at this point, but feel free to reopen if I'm misreading and/or you disagree

closed 2021-02-19 by larsoner