mne-tools/mne-python #5124: BUG- can no longer compute projs from CTF data

bloyl, 2018-04-11:
PR #5083 added an assert to make sure that ref channels are retained when picking data channels. this had the side effect of breaking EOG and ECG proj computations.

reference channels are explicitly not included when computing proj operators (https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ssp.py#L186-L187). However there is no mechanism to tell pick_drop_channels to not care about the compensation channels.

here is the traceback
> File "<string>", line 2, in compute_proj_eog
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/utils.py", line 729, in verbose
>     return function(*args, **kwargs)
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/preprocessing/ssp.py", line 406, in compute_proj_eog
>     iir_params=iir_params, return_drop_log=return_drop_log)
>   File "<string>", line 2, in _compute_exg_proj
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/utils.py", line 729, in verbose
>     return function(*args, **kwargs)
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/preprocessing/ssp.py", line 202, in _compute_exg_proj
>     evoked = epochs.average()
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/epochs.py", line 824, in average
>     return self._compute_mean_or_stderr(picks, 'ave')
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/epochs.py", line 893, in _compute_mean_or_stderr
>     kind, self._name)
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/epochs.py", line 923, in _evoked_from_epoch_data
>     evoked.pick_channels(ch_names)
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/channels/channels.py", line 716, in pick_channels
>     pick_channels(self.info['ch_names'], ch_names))
>   File "/mnt/isilon/meg_lab/bloyl/Programs/mne-python-devel/mne/channels/channels.py", line 815, in _pick_drop_channels
>     assert len(comp_picks) == len(comp_names)
> AssertionError

larsoner, 2018-04-11:
> However there is no mechanism to tell pick_drop_channels to not care about the compensation channels.

You can pass `check_comps=False` to the private `_pick_drop_channels`:

https://github.com/mne-tools/mne-python/blob/master/mne/channels/channels.py#L796

Can you see if the function can be modified to use this? In particular doing `Epochs(... picks)` might not be the way to go, but rather create the object, then use the private drop function with `check_comps=False` to do the picking.

bloyl, 2018-04-11:
I see two options:

1. Since `_compute_proj` currently rejects the `ref_meg` channels at computation time:
https://github.com/mne-tools/mne-python/blob/master/mne/proj.py#L72-L76
we could keep the ref_meg channels in the epochs object and just let `_compute_proj` take care of removing them?

2. In `_compute_exg_proj` we could test if the supplied raw is an instance of RawCTF and only keep the ref channels for those instances.

larsoner, 2018-04-11:
There is a third option (that I hinted at), don't do picking with the `Epochs` constructor at all but immediately afterward with a call to the private `epochs._pick_drop_channels(picks)`. This should effectively work out to be a worse version of your (1) if `_compute_proj` is smart enough to leave the reference channels out during its computation. If it isn't, though (if it includes the ref MEG sensors), then my proposal might be easiest, or we can make the `_compute_proj` smarter (up to you).

In general we should not do special cases for CTF data, it will lead to breakages if `comps` is ever used by another system (future compat) or for another purpose (e.g., I have thought about how you could store any arbitrary linear spatial transform in there).

PR #5125 (2018-04-22): FIX: CTF Projector comp
Closes #5124 

When computing ssp projection operators, include the meg_ref channels for data objects that include compensation channels (i.e CTF data), otherwise creating epochs raises an exception.
closed 2018-04-22 by larsoner