mne-tools/mne-python #5127: Strange behavior generating epochs with gradient compenstation on

bloyl, 2018-04-12:
There is a potential issue generating epochs with gradient compensation. Following PR #5083 the expectation would be that reference channels would be required generating epochs from data with compensation matrices. However that isn't the case. 

The following code demonstrates the strange behavior see the comments for my expected behavior.

```
import os.path as op

from mne.datasets import testing
from mne.io import read_raw_ctf
from mne.io.pick import pick_types
from mne import Epochs
from mne.event import make_fixed_length_events

ctf_fname = op.join(testing.data_path(download=False), 'CTF', 'testdata_ctf.ds')

raw = read_raw_ctf(ctf_fname, preload=True)

# make sure gradient compensation is on
assert(len(raw.info['comps']) > 0)  # has comp channels
raw.apply_gradient_compensation(3)

# try to pick channels without references
picks = pick_types(raw.info, meg=True, eeg=False, eog=False, ref_meg=False)
ch_names_picked = [raw.ch_names[idx] for idx in picks]

events = make_fixed_length_events(raw, 99999)

# would expect this to fail but it doesn't particularly with preload=True
epochs = Epochs(raw, events, None, -0.2, 0.2, preload=True, picks=picks)

# make sure we have all the channels
assert(len(set(epochs.ch_names).difference(ch_names_picked)) == 0)
assert(len(set(epochs.info['ch_names']).difference(ch_names_picked)) == 0)
# so all the info in the epochs object suggest we have the chans we asked for.

# A user now has access to incorrect data?
assert(epochs.get_data().shape[1] == len(epochs.info['ch_names']))

# something is keeping track of this because this fails.
epochs.average()
```

larsoner, 2018-04-12:
I probably put the check for `comps` in the wrong place (in `channels.py` for `_pick_drop_channels`). If we move the check into `pick_info` (with new default arg `check_comps=True`) it should also make the `epochs` line error.

Can you see if it is indeed the case?

bloyl, 2018-04-12:
I don't totally follow what you are suggesting.

Although, in looking through `pick_info`, I wondered if we should add the `comps` check to `_check_consistency()` of `meas_info.py`. 

I would say that an info is invalid if `compensation_grade != 0` and there are compensation channels that don't exist in the info.

Does that seem like a reasonable check?

larsoner, 2018-04-12:
I would say the steps are:

1. Add `check_comps=True` kwarg to `pick_info`
2. Move `comps`-checking code from `mne.channels.UpdateChannelsMixin._pick_drop_channels` to `mne.io.meas_info.py` as a new function `_bad_picks_comps(info, picks)` or so that returns `True` if the picks are "illegal" and False if they are okay.
3. Call this new checking function in `pick_info` and in `_check_consistency` and raise an error in each place if it returns True. The reason to do it in `pick_info` in addition to `_check_consistency` is that the error messages should be different. In `pick_info` we can raise a clear error about the set of `picks` being invalid. In `_check_consistency` we don't know it's necessarily due to a user doing a `picks` operation or something else such as a reading error (this call is used a lot in `mne` and not necessarily just in `pick_*` functions) so the error message should be generic. Maybe we will never violate the check in `_check_consistency` but that's okay.

larsoner, 2018-04-12:
> I would say that an info is invalid if compensation_grade != 0 and there are compensation channels that don't exist in the info.

As for the logic, this should be what you find already in `mne.channels.UpdateChannelsMixin._pick_drop_channels` in `master`

larsoner, 2018-04-12:
See:

https://github.com/mne-tools/mne-python/blob/f5e3399ad6d0c163f3b86f002fd3c6fb687e6315/mne/channels/channels.py#L808

PR #5133 (2018-04-30): MRG+1: check info against compensation chans
@larsoner, take a look at this.

The main difference from what we discussed is that I removed the check_comp flags, as that mechanism seemed like something that could start to be abused and would lead to headaches later.

Closes #5127
closed 2018-04-30 by agramfort