mne-tools/mne-python #5128: cross_val_multiscore: y_true/y_pred?

giuliagennari, 2018-04-13:
Hello,

I was wondering whether there is a way to access the actual predictions other than the scores when using this. If not would you have any suggestion on the best way to do that? 
I can not use  sklearn.model_selection.cross_val_predict since my cv scheme is StratifiedShuffleSplit(100).
Is an interation over my cv splits with estimator.fit and estimator.predict the only option? It seems very unconcevient....

Thank you very much in advance!!!!

jona-sassenhagen, 2018-07-30:
Essentially a duplicate of #4606, right?

We should probably eventually get `cross_val_multipredict`, which is what @giuliagennari is asking for, I think.

agramfort, 2018-07-30:
I am fine coding the equivalent of http://scikit-learn.org/stable/modules/generated/sklearn.model_selection.cross_val_predict.html

agramfort, 2018-08-04:
yes dupe of #4606

closed 2018-08-04 by agramfort