mne-tools/mne-python #5138: Simulation with simulate_raw and labels

mbaumont, 2018-04-17:
Hi, 
I have just started to work with python and MNE for a project. I am making a MEG/EEG simulator in order to test signal processing methods, and therefore I found the various tools made available on MNE greatly useful. For my project I made a small graphical interface to choose different sources on the surface of the brain, and I save those sources with the label class. I then make a list with those labels and use mne.simulation.simulate_stc and simulate_raw to do the simulation. This worked fine unless only one label was present in my label list, or if the sources were only present in one hemisphere.  
It gives me an error in mne\simulation\raw.py in _restrict_source_space_to :
```
s['inuse'][s['vertno']] = 1
IndexError: arrays used as indices must be of integer (or boolean) type
```
It seems to be because if there are no sources in one of the two hemispheres, stc.vertices still gives an empty array for the hemisphere. I added line 542 of \mne\simulation\raw:
```ruby
if len(v):
        s['inuse'][s['vertno']] = 1
```

And now the simulation works fine. I am not sure if this is a real bug or just something I did wrong with my simulation.

agramfort, 2018-04-17:
@mbaumont can you provide a full code snippet to replicate the pb using the sample dataset?

larsoner, 2018-04-17:
Somehow `s['vertno']` must be `np.array([])` instead of `np.array([], int)`. So I suspect somewhere in your code or ours there is a `np.array(...)` where there needs to be an `np.array(..., int)`. In cases where the `...` contains integers, numpy automatically creates an integer array. But when `...` is an empty list, it can't infer the type as `int` so it uses `float`.

mbaumont, 2018-04-17:
Indeed my stc has a rh_vertno (or lh_vertno depending on which hemisphere has no sources) of type `np.array([], float)`. If I set the empty vertices to be of type int  (`stc.vertices[1] = stc.vertices[1].astype(int)` ) the code below works. 

```ruby
import mne 
from mne.datasets import sample
import numpy as np 

data_path = sample.data_path()

raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
trans_fname = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'
src_fname = data_path + '/subjects/sample/bem/sample-oct-6-src.fif'
bem_fname = (data_path + 
                    '/subjects/sample/bem/sample-5120-5120-5120-bem-sol.fif')

raw = mne.io.read_raw_fif(raw_fname)
raw = raw.crop(0., 30.)  # 30 sec is enough

epoch_duration = 2.  # duration of each epoch/event
times = raw.times[:int(raw.info['sfreq'] * epoch_duration)]

def data_fun(times, n_dipoles, time_course):
    data =  np.zeros((n_dipoles, len(times)))
    rng = np.random.RandomState(42)
    for i in range(n_dipoles):
        times_simu = times - time_course[i]
        data[i] = (50e-9 * np.sin(30. * times_simu + 0.2) *np.exp(- (times_simu
                                    - 0.15 + 0.05 * rng.randn(1)) ** 2 / 0.01))
    return data

stc_data = data_fun(times,1,[0])
stc_data.reshape((-1, 1))

labels = list()
label_fname = data_path + '/MEG/sample/labels/Aud-lh.label'
label = mne.read_label(label_fname, subject=None, color=None)
labels.append(label)
tstep = 1/raw.info['sfreq']

src = mne.read_source_spaces(src_fname)
stc = mne.simulation.simulate_stc(src, labels, stc_data, 0,
                                  tstep)
#stc.vertices[1] = stc.vertices[1].astype(int)

raw_sim = mne.simulation.simulate_raw(raw, stc, trans_fname, src, bem_fname, 
                       cov='simple',iir_filter=[0.2, -0.2, 0.04], ecg=True, 
                       blink=True)
```

larsoner, 2018-04-17:
Ah you create it with `mne.simulation.simulate_stc`, so it is probably a bug there. I'll take a look

larsoner, 2018-04-17:
@mbaumont this bug has already been fixed on master in https://github.com/mne-tools/mne-python/commit/e744aff13463e32ab719f1eee74c764cdac8bcc4 so I'll close, but please feel free to check to make sure!

closed 2018-04-17 by larsoner