mne-tools/mne-python #5142: mne.io.read_raw_brainvision reads incorrect number of datapoints

ArcCha, 2018-04-18:
I am writing a simple time series visualization script basing on MNE. I need to read data from EDF and BrainVision txt files, I have a sample that was exported to both. I discovered a strange behaviour that probably is a bug in `mne.io.read_raw_brainvision`, it is illustrated by

```
#!/usr/bin/env python
import mne

edf = mne.io.read_raw_edf("datasets/example.edf")
bv = mne.io.read_raw_brainvision("datasets/dattxt/example.vhdr")

print("Number of datapoints per channel ", edf.n_times, bv.n_times)
```

which prints in my case `('Number of datapoints per channel ', 831488, 73)`. It can be found (together with sample data) [in this repo](https://github.com/ociepkam/pfb). I would expect that `edf.n_times` and `bv.n_times` be equal. 

Additionally, when I try to read data from `bv` like so

```
data, time = bv[:, :1000]
```

I get

```
Traceback (most recent call last):
  File "./test.py", line 9, in <module>
    data, time = bv[:, :1000]
  File "/home/arccha/.conda/envs/pfb/lib/python2.7/site-packages/mne/io/base.py", line 841, in __getitem__
    verbose=self.verbose)
  File "/home/arccha/.conda/envs/pfb/lib/python2.7/site-packages/mne/io/base.py", line 523, in _read_segment
    cals, mult)
  File "/home/arccha/.conda/envs/pfb/lib/python2.7/site-packages/mne/io/brainvision/brainvision.py", line 116, in _read_segment_file
    _read_segments_c(self, data, idx, fi, start, stop, cals, mult)
  File "/home/arccha/.conda/envs/pfb/lib/python2.7/site-packages/mne/io/brainvision/brainvision.py", line 174, in _read_segments_c
    dtype = _fmt_dtype_dict[raw.orig_format]
TypeError: unhashable type: 'dict'
```

Am I doing something wrong or is it indeed a bug?

cbrnr, 2018-04-18:
I can't read those files at all - maybe something went wrong when you put them in the repo? Can you share these files e.g. via Dropbox?

ArcCha, 2018-04-18:
@cbrnr as this files are rather big I've used [GIT LFS](https://git-lfs.github.com/) to put them in the repository. You need to have it installed in order to clone them.

cbrnr, 2018-04-18:
I can't load the BrainVision file because the `.vhdr` file contains invalid data and marker file names. When I set these fields to `example.dat` and `example.vmkr`, I also get only 73 samples. However, I also cannot load this file properly with SigViewer (I don't see any signals, just markers). How did you create this BrainVision file? Can you open it with e.g. BrainVision Analyzer?

ArcCha, 2018-04-18:
Yes, you’re correct, the `.vhdr` file contained wrong filenames. I’ve fixed it. I was able to load this file sucessfully with BVA 2.1.2.327. It was generated by exporting previously captured file using BVA export option.

cbrnr, 2018-04-19:
I can reproduce the error. I could successfully load this file with EEGLAB and the FileIO plugin, but this took a long time (and produced a warning for each channel: `unknown resolution (i.e. recording units) for channel 73 in example.vhdr`), but the signals look OK. The BioSig toolbox for MATLAB also takes ages, and at some point I aborted the process so I'm not sure if it would have worked.

So yes, it seems like there is a bug in our BrainVision code. I'll look into it when I have more time.

bethard, 2018-06-07:
I'm ran into the same issue. In case it helps, the `TypeError` seems to be caused by `raw.orig_format` being a dictionary:
```python
>>> data.orig_format
{'decimalsymbol': '.', 'skiplines': '0', 'skipcolumns': '1'}
```
I looked into the vhdr file, and I imagine this is coming from:
```
[ASCII Infos]
; Decimal symbol for floating point numbers: the header file always uses a dot (.),
; however the data file might use a different one
DecimalSymbol=.
; SkipLines, SkipColumns: leading lines and columns with additional information.
SkipLines=0
SkipColumns=1
```
That apparently happens if we export from BrainVision in text format. If we switch to binary format, then the vhdr file instead contains:
```
[Binary Infos]
BinaryFormat=IEEE_FLOAT_32
```
And `read_raw_brainvision` loads the file without the `TypeError`.

agramfort, 2018-06-07:
either we raise a meaningful error or we make it work...

bethard, 2018-06-07:
+1 on raising the error and filing a separate feature request for support of the ASCII flavor. A clearer error message explaining what the problem was would have definitely saved some time, and exporting in BINARY instead wasn't a big deal, at least in my situation.

agramfort, 2018-06-07:
ok sounds like a plan

can you open a PR?

PR #5271 (2018-06-14): [MRG] Raise NotImplementedError for vectorized Brainvision ASCII data file
Fixes #5142.
closed 2018-06-14 by larsoner