mne-tools/mne-python #5144: Spatiotemporal permutation cluster test for time frequency data?

Yugga, 2018-04-18:
Hi, 

Mne.stats has some excellent permutation functions, but I'm missing one feature there:
-For mne.stats.permutation_cluster_test, it is possible to give data input as a 3-dimensional time-frequency power representation
-For mne.stats.spatio_temporal_cluster_test, I can give the whole data set with all the channels, and get some statistics concerning spatial relationships.

However, I haven't found a way to combine these, i.e., have the input as (n_observations, n_channels, n_frequencies, n_times). I'm interested in comparing two sets of time-frequency data, and would like to know in which sensors, in which frequencies and in which time points the differences between the two are significant. 

Is this possible, and how do you set up the connectivity matrix in that case? 

In the Tutorial examples https://www.martinos.org/mne/stable/auto_tutorials/plot_stats_cluster_time_frequency.html and https://www.martinos.org/mne/stable/auto_tutorials/plot_stats_cluster_1samp_test_time_frequency.html, one has restricted the examination to a single channel level, but would it be possible to extend this to all the channels?

larsoner, 2018-04-18:
For now you will need to reshape it to be `(n_observations, n_channels * n_frequencies, n_times)`, and then construct a "connectivity" matrix that is correct for the `n_channels * n_frequencies` dimension. To define the connections between frequencies, this would be a block diagonal of `n_channels` copies of these:
```
np.eye(n_frequencies) + np.eye(n_frequencies, k=1) + np.eye(n_frequencies, k=-1)
```
Then you'd need to add the channel connectivity to this matrix, as well. Someone will have to take the time to describe how to do it and work it out, but if you think about the dimensions it should be doable starting from a `(n_channels, n_channels)` connectivity matrix.

If someone can flesh this out, we can add a function to compute such a matrix, or at least make an example of it for now.

FYI we are discussing a cleaner API to eventually use here:

https://github.com/mne-tools/mne-python/issues/4859

christianbrodbeck, 2018-04-18:
In eelbrain this should work out of the box if you want to try it (http://eelbrain.readthedocs.io/en/r-0.27/reference.html#module-testnd); to use MNE objects: http://eelbrain.readthedocs.io/en/r-0.27/generated/eelbrain.load.fiff.html

Yugga, 2018-04-22:
Excellent, thanks for the advice! I'll see if I can figure this out.

larsoner, 2018-04-23:
And thus #4859, let's continue discussion there

closed 2018-04-23 by larsoner

larsoner, 2024-08-01:
This appears to be a question about usage, not a bug report or a new feature request. Please use the [MNE forum](https://mne.discourse.group/) for usage questions.