mne-tools/mne-python #5148: BUG?: average=False in compute_proj_exg

larsoner, 2018-04-20:
By default we have `average=False` in `compute_proj_ecg` and `compute_proj_eog`. But when I was toying with the idea of using `Xdawn` (which I think might still be a good idea) to compute projectors rather than plain SSP, I found that `average=True` seemed to always be better than `average=False`.

Here is the sample data, with four columns: the first two are `average=False`, the last two are `average=True`; within those, the left are ECG epochs (to be projected) and the right are auditory epochs (hopefully not contaminated):

[image]

Note that the "average=True" does not really make sense for Xdawn because it already uses the average, so the first two and last two columns are the same for Xdawn.

I think we might consider:

1. Changing the default `average=False` to `average=True`.
2. Probably unify `compute_exg_proj` and `create_exg_epochs`. The former does not use the latter but it probably should.
3. Implement Xdawn as a backend instead of just SSP. Even with `average=True` it should do better against other noise present in the epochs, especially if that noise is large (yet to be shown).

larsoner, 2018-04-22:
I suggest we use a deprecation cycle to get to `average=True`, then, for 0.16. The others can be 0.17.

agramfort, 2018-04-22:
Ok for me

cbrnr, 2018-04-23:
+1 for tackling `average=True` now and the rest later. But why do you think this is a bug?

larsoner, 2018-04-23:
It probably should have been True from the beginning.

>

cbrnr, 2018-04-23:
So results are wrong if `average=False`?

larsoner, 2018-04-23:
They seem to be less correct in testing.

cbrnr, 2018-04-23:
OK, so probably not a but, but suboptimal parameter settings.

massich, 2018-04-30:
I don't think so (because point 3 is missing), but we can take it off of the 0.16 milestones.

agramfort, 2018-04-30:
ok done

larsoner, 2018-10-04:
Point (3) is open we just need to decide about whether or not to do it

closed 2018-10-04 by larsoner