mne-tools/mne-python #5151: turn off evoked Nave printing?

jona-sassenhagen, 2018-04-21:
Shouldn't there be an option to turn off Nave printing? Otherwise some forms of plots can look a bit awkward:

[image]

@larsoner

larsoner, 2018-04-22:
ax.texts = []

Good enough?

>

jona-sassenhagen, 2018-04-22:
Hm, not really, cause there might be a ton of other texts there right?

So far, I'm setting `evoked.nave = None`, but that does not seem right.

larsoner, 2018-04-23:
Ahh I thought there was 1 but there are 3:
```
>>> mne.read_evokeds(mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis-ave.fif')[0].plot().axes[0].texts
Text(0, 0, 'blank'),
 Text(0, 5, 'N$_{\\mathrm{ave}}$=55'),
 Text(0.01, 0.1, 'Loading...')]
```
So `ax.texts.pop(1)` would do it.

I'd rather not clutter the API with another option. Can we instead think of a better placement so that removal is not necessary? Maybe right-aligned to the Y-axis ?

jona-sassenhagen, 2018-04-27:
Yeah another kwarg seems overkill ... I'm not sure.

larsoner, 2019-08-20:
I'm going to close this since I seem to remember us having some suitable solution to this in a PR, but @jona-sassenhagen feel free to reopen if it's still an issue

closed 2019-08-20 by larsoner