mne-tools/mne-python #5152: axes reversed on coreg

jshanna100, 2018-04-22:
The MNE I'm running is the latest development version as of this afternoon, running in the conda environment as specified here: https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml

The axes in mne.gui_coregistration() are reversed (left is right, top is bottom).

jshanna100, 2018-04-23:
I believe it's opaque. The fiducials are also invisible from most angles, and changing them by click doesn't seem to work with any reliability. MNE 15 seems to have none of these problems on the same data/environment.

[image]

jshanna100, 2018-04-23:
Also, altering the viewpoint with the mouse is reversed, i.e. dragging right rotates in the opposite direction.

larsoner, 2018-04-23:
Can you do "display options" and ensure that the head is set to `opacity=1.`. Does it change when tweaking the opacity in the range between 0 and 1?

All of the behaviors you describe are consistent with the head having opacity `0.`, as the inside surface always has opacity `1.`.

jshanna100, 2018-04-23:
It was set to 1, but something interesting happened when tweaking it: it
does indeed return to normal at values very near 1, but 1 itself breaks it.

larsoner, 2018-04-23:
Ahh okay. I have seen this sort of bug with Mayavi before. I'll see if I can make a PR to fix it

PR #5157 (2018-04-30): MRG: Fix ordering
Closes #5152.

@jshanna100 can you see if this fixes the problem for you?
closed 2018-04-30 by agramfort