mne-tools/mne-python #5154: Switch default colors to MPL defaults?

jona-sassenhagen, 2018-04-23:
Since 2.0, the MPL defaults (c0, c1, ...) are very good. Should we not transition there with, for example, our line plot defaults?

larsoner, 2018-04-23:
Which line plots do you have in mind?

jona-sassenhagen, 2018-04-23:
All plots that use mne.viz.utils.COLORS, i.e., the annotation plots.

larsoner, 2018-04-23:
Sounds good to me. I think we can just change these and put a note in the API section.

sappelhoff, 2018-05-04:
See PR .. is this what you meant @jona-sassenhagen? [Here](https://matplotlib.org/users/dflt_style_changes.html#colors-in-default-property-cycle) are the colors.

However, this change means going from 
```python
['b', 'g', 'r', 'c', 'm', 'y', 'k', '#473C8B', '#458B74',
 '#CD7F32', '#FF4040', '#ADFF2F', '#8E2323', '#FF1493']
```

to 
```python
[u'#1f77b4', u'#ff7f0e', u'#2ca02c', u'#d62728',
 u'#9467bd', u'#8c564b', u'#e377c2', u'#7f7f7f', u'#bcbd22', u'#17becf']
```

i.e., fewer colors. Is that an issue?

jona-sassenhagen, 2018-05-04:
We could just stick with `'c0', 'c1', 'c2', ...` right?

larsoner, 2018-05-04:
And we do have conditionals that look for `'r'` in the color cycle somewhere in the viz plotting code having to do with annotations. So that will need to be changed.

PR #5186 (2018-08-12): [MRG] use mpl default color cycle
fixes #5154
closed 2018-08-12 by larsoner