mne-tools/mne-python #5159: Can't read events from BDF file

ElBibulo, 2018-04-24:
Hi all,

When i import my data from the BDF format ([test.zip](https://github.com/mne-tools/mne-python/files/1941601/test.zip)) i can't read the event list. The events are present in the data as i can read them using EEGLAB.

i loaded the code using:
`raw = mne.io.read_raw_edf(bdf_file, stim_channel="Status")`
i have tried
`raw.find_edf_events()`
as well as
`mne.find_events(raw)`

both functions return an empty array.

when i use `raw.plot()` the stimulus channell appears to be completely flat.
i looked at the stim channel value for each sample:
`data = raw.to_data_frame()`
`print(data["STI 014"])`
the value at every sample is 41009.0 ( i have no idea where that comes from)

Not sure where i am doing something wrong (I'm pretty new to MNE-Python and EEG-Data processing in general) and i hope you can help me with this.

best,
Ole

cbrnr, 2018-04-24:
I can confirm this issue. It has something to do how we (automatically) convert the stim channel. If you don't convert it, you actually get this picture:
[image]
I will look into it.

cbrnr, 2018-04-24:
Also, it is OK that `raw.find_edf_events()` doesn't return any events, because your data file does not have a dedicated annotation channel. Instead, events are stored in the last channel as a regular data channel.

cbrnr, 2018-04-24:
I think the problem is that the last channel gets scaled, because its physical range is 374940 and its digital range is 16777215. Each channel get scaled by the quotient, which in this case is 374940 / 16777215 = 0.022348166844139507 (just like for all other 24 channels in this file). Normally, the stimulus channel has identical physical and digital ranges, which makes the scaling factor equal 1.

Now apparently MNE relies on the channel scaling to extract events. One way to fix this is to enforce a scaling factor of 1 for stim channels. However, I don't know if this breaks other things, but it might be worth a try. At least this works with your test file.

Do you expect specific event types (numbers) for the three types?

ElBibulo, 2018-04-24:
Great, now i can get the eventlist. As is understood i have to subtract the values in the third column of the event list from the 'baseline' in the second. If i do this i get event types 1,2 and 4 which is what i am expecting. Thank you!

PR #5160 (2018-06-27): WIP: Set physical range to digital range for stim channels
Fixes #5159. This is a long shot, let's see if other things break.
closed 2018-06-27 by larsoner