mne-tools/mne-python #5161: problems installing mne from ubuntu 14.04 terminal

ShirsankarBasu, 2018-04-24:
I'm currently using ubuntu 14.04, and trying to install mne library from the ubuntu terminal, invoking commands from the following link:
 https://www.martinos.org/mne/stable/install_mne_python.html.
But, it's not getting done. Here, for you to debug, I'm giving a flowchart, and also attaching a screenshot of the error messages being issued against the commands I typed. The version of my operating system is 14.04, with conda, pip and python already installed.

A step-by-step flowchart of how I'm trying to install mne from ubuntu terminal:

step-1: open the terminal.
step-2: go to the directory where anaconda environment is installed invoking '$ cd anaconda3' command.
step-3: Invoking the following command from terminal:
           ($ conda install scipy matplotlib scikit-learn mayavi jupyter spyder)
step-4: Invoking the second command from ubuntu terminal:
           ($pip install PySurfer mne)

larsoner, 2018-04-24:
Can you try our more up-to-date instructions here?

https://martinos.org/mne/dev/install_mne_python.html

ShirsankarBasu, 2018-04-24:
Thanks. But, first of all-- I think--to make this command('$curl -O https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml') work from terminal, I need to have 'curl' already installed in my operating system. But, I faced problems in installing curl as well. Curl is not getting installed. Can you please give any lead to that?

agramfort, 2018-04-24:
what OS are you using?

you can just download/save the file
https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
from your browser if you don't have curl.

ShirsankarBasu, 2018-04-25:
Im using ubuntu 14.04. I tried installing curl from command line of ubuntu operating system, from the following link: http://devendraverma.com/install-curl-on-ubuntu-14-04/
but, it's not working. Please, tell me how to install curl as to install mne from ubuntu terminal.

Thanks.

agramfort, 2018-04-26:
`sudo apt-get install curl` should work.

closed 2018-04-26 by agramfort