mne-tools/mne-python #5163: minor issue: MNE naming convention enforcement

sappelhoff, 2018-04-25:
I would like to save my ICAs in a "BIDS-like" fashion, such as `sub-002_task-bandit_ica.fif.gz`, where "bandit" is the task ... However, this triggers a warning from `check_fname`:

https://github.com/mne-tools/mne-python/blob/9865f9275f596443668a06e84e54f2e2c645bddb/mne/preprocessing/ica.py#L1385

While I see the benefit of using `ica` at the end of the file name, is it crucial to enforce `-ica` as the convention?  I suggest to drop the `-` before `ica`.

cbrnr, 2018-04-25:
I made this suggestion in #4813, but people seemed to disagree on this topic. I still think a good solution is to enforce the `-ica` at the end, but instead of a warning just issue a message at the info level.

sappelhoff, 2018-04-25:
Having read that issue and considering your list:

> 

    1 Everything is fine, stop yelling about that and just rename your files.
    2 Messages related to violating file naming conventions should be in the info level and not warnings.
    3 Let's just include ..._ica.fif in the list of acceptable file names.
    4 Let's downgrade the warnings to infos in read_ica, but let's leave them as warnings in ica.save.
    5 If these warnings bother you, use read_ica(..., verbose='error').
    6 Let's have a way to disable specific warnings in the use_log_level context manager.

I'd go with 2 as well. Feel free to close this if all of you have had enough of this discussion. But the BIDS argument is on the side of the '_' convention over '-' :-)

larsoner, 2018-04-25:
@cbrnr must have had some clairvoyance about `-` vs `_` :)

I'm fine with extending these to `_ica` variants. I think even @agramfort will agree given it's BIDS-like (assuming it is). @jasmainak should we add `-whatever.fif` to `_whatever.fif` to our file checks?

As for info vs warning can we assume we don't need to change it if `_` is permitted?

sappelhoff, 2018-04-25:
As for this issue, definitely ... I just want to BIDSify my file names, which is possible with `_` and not get the warnings by MNE.

larsoner, 2018-04-25:
@sappelhoff do you have a list of changes that you need to make and what warnings they give you on `master` so far? A PR to extend our permitted filenames, even if it's not 100% complete, would be worthwhile and a potentially easy thing to get in for 0.16 (which we are trying to push out this week).

sappelhoff, 2018-04-25:
Mmh not sure what you mean, so far on my list, the only change I'd like to make would be to change

https://github.com/mne-tools/mne-python/blob/9865f9275f596443668a06e84e54f2e2c645bddb/mne/preprocessing/ica.py#L1385

to:

`check_fname(fname, 'ICA', ('-ica.fif', '-ica.fif.gz', '_ica.fif', '_ica.fif.gz')) `

if we want to allow `_` for all `check_fname` calls, then I'd first have to find them all ...

sappelhoff, 2018-04-25:
But I think a PR on that would be rather quick and painless, and adding tests shouldn't be a problem either - if required at all?

larsoner, 2018-04-25:
I mean if you also had problems saving `Evoked` or `Epochs` in BIDS-compatible fashion, we should change those, too.

These instances are easy to find with a sufficient `git grep` regular expression :)

As for tests, I think we already check for violations, strict TDD would dictate you add one that `_ica.fif` is acceptable but I wouldn't make you do it :)

Let's wait to see what @agramfort and @jasmainak think about making an incremental `_ica.fif` PR or if they have a complete list we can do all at once (or none at all if they have some other plan).

sappelhoff, 2018-04-25:
Or we could use the "search this repository" function by Github? [see here](https://github.com/mne-tools/mne-python/search?utf8=%E2%9C%93&q=check_fname&type=)

larsoner, 2018-04-25:
From chatting with @jasmainak, the BIDS naming convention uses `<key1>-<value1>_<key2>-<value2>..._modality.extension`. Underscore is used to separate key-value pairs, so `_ica.fif` and `-ica.fif` are both incompatible.

There will be a meeting in August to discuss derivative filenames like you'd use for ICA for example. My sense is that we should scrap the filename check altogether (no `warn` or `logger.info`), aside from looking for `.fif` or `.fif.gz` extensions for now. Otherwise we are pushing people toward a standard that will be obsolete soon.

The one place this is problematic is `Report`, which expects the current `-ave.fif`-style filenames. I think we can add to the `Notes` section of `Report` to say what it currently will parse properly, and update it once the BIDS standard is fully usable.

@jasmainak said he's fine with this, @sappelhoff @cbrnr it sounds like from above/before you agree with eliminating the warning (though perhaps not putting it back eventually, but we can decide that later). Let's see if @agramfort agrees.

cbrnr, 2018-04-25:
This sounds great! Let's scrap the warnings, or alternatively demote them to info.

teonbrooks, 2018-04-25:
BIDS hasn't conventionalized derivative files yet, just raw (or minimally treated), continuous files. we are working on BIDS-ifying the output from mne with [mne-bids](https://github.com/mne-tools/mne-bids).

agramfort, 2018-04-26:
as I said before, no objection but it should be a global decision. either we don't have warning anywhere or we do. But this is a longer shot than just changing:

`check_fname(fname, 'ICA', ('-ica.fif', '-ica.fif.gz')) `

to:

`check_fname(fname, 'ICA', ('-ica.fif', '-ica.fif.gz', '_ica.fif', '_ica.fif.gz'))`

which I can easily accept at it's already the case for Raw.

cbrnr, 2018-04-26:
Then let's add underscores to the list of accepted file names (not just for ICA).

agramfort, 2018-04-26:
ok

sappelhoff, 2018-04-26:
@cbrnr are you making a PR or should I?

cbrnr, 2018-04-26:
I'll do it, see #5166.

larsoner, 2018-04-26:
I think we should get rid of these checks (get rid of them everywhere). We
are enforcing an obsolete or nearly obsolete standard.

>

larsoner, 2018-04-26:
... but at least `_` is incremental progress, so we can do that for now

PR #5166 (2018-04-26): [MRG] Add _ to valid file names
Fixes #5163.
closed 2018-04-26 by larsoner