mne-tools/mne-python #5167: Python 2 support roadmap

cbrnr, 2018-04-26:
What's the roadmap towards sunsetting Python 2 support for MNE? Many projects have already decided to support only Python 3, with one final Python 2 release that won't get any new features. MNE should be on the http://python3statement.org/ website, which means that we will have to drop Python 2 support by 2020 at the latest. Many projects will do or have done this earlier by releasing one final Python 2-compatible version, while development continues with Python 3-only code.

larsoner, 2018-04-26:
My vote is to follow NumPy and SciPy:

https://docs.scipy.org/doc/numpy-1.14.0/neps/dropping-python2.7-proposal.html

But for use our maintenance releases don't last nearly as long as theirs. So the important thing is that we drop new features at the end of 2018, which means 0.17 would probably be the last 2.7 release.

jona-sassenhagen, 2018-04-27:
>My vote is to follow NumPy and SciPy

Yeah, why not?

cbrnr, 2018-05-02:
I like the idea of following NumPy. Should we prepare a short statement for the website? Should we also try to get on the Python 3 Statement website?

@agramfort it's nice that we currently support the two versions, but at least for me it is sometimes annoying/frustrating to keep stumbling over Python 2 incompatibilities (related to e.g. encoding or float/int). Even more so, we can't use some nice Python 3 features and have to carry around the six package to maintain Python 2 compatibility. I think discarding all that will free our resources for getting more important things done. The whole CI suite is already taking a long time, so dropping all Python 2 jobs will be a welcome relief for me (and Travis).

The other question is: does everything work with Python 3? I remember some people mentioning a while ago that mayavi is still in beta for Python 3, but this seems to be resolved.

How is this going to work technically? After we make the last Python 2 release (presumably 0.17), do we actively start to make the code more Python 3 like, or do we not touch it and just ensure that new features make use of Python 3? I'm assuming the latter, and the former only if someone has time to do it.

agramfort, 2018-05-02:
if for my next MNE training course all students have a working python 3 env
I am happy to make this decision rapidly.

we should ask on the mailing list if there is any objection too.

cbrnr, 2018-05-02:
I think this should be possible since installing Python 3 via Anaconda is not more difficult than installing Python 2 :smile:. I'll ask on the mailing list.

jona-sassenhagen, 2018-05-02:
We probably want to hear more MNE devs here. @kingjr in particular :)

cbrnr, 2018-05-02:
Yes, more input would be great. I've already sent out a message to the mailing list (although with the wrong email address so this message is awaiting moderator approval).

kingjr, 2018-05-02:
How s mayavi on python 3 ?

jona-sassenhagen, 2018-05-02:
@larsoner claims if you use the environment.yml to set up your system, it works. Haven't tried it myself, but if Eric says so ...

cbrnr, 2018-05-02:
Also https://github.com/enthought/mayavi/issues/84 is closed, so I'm assuming that it works (but haven't tried it either).

larsoner, 2018-05-02:
For almost 7 months our CIs have been running Py3k with Mayavi.

kingjr, 2018-05-02:
Do they include Mac and windows environments ?

larsoner, 2018-05-02:
Yes

christianbrodbeck, 2018-05-02:
Ideally the dependencies would be available as official release rather than special github branch installations... Hopefully that will be the case soon...

palday, 2018-05-02:
I'm also all for dropping explicit Python 2 support once all the dependencies have official Py3-compatible releases and don't require any extra steps to install from a development branch. 

How dependent is the overall codebase on `six` for dual compatibility? My own experience is that well-written code, even string code, generally works well[1] on 2.7 and 3 without any changes except a `from __future__ import print_function, division` (at least for code that's not dependent on features introduced in a later Py3 point release like the new format strings). If that's the case, then we can leave that line in, even after we drop official Py2 support / CI testing and let limited, unsupported Py2 compatibility continue until too much of the code has Python 3 idioms.

[1] This of course ignores some memory optimality for some iterator methods.

larsoner, 2018-05-02:
We use `six` for a bunch of things, like `next`, `StringIO`, `BytesIO`, `string_types`, etc.

cbrnr, 2018-05-03:
@palday which dependencies do not have official Python 3 support? AFAIK all packages in `environment.yml` are officially on PyPI, so it might not be necessary to pull them from specific branches.

palday, 2018-05-05:
I didn't have anything in particular in mind and I haven't looked at the Install page in a while, but I was thinking about how there used to be some special casing for Python 3, which I think is what @christianbrodbeck was also getting at.  My statement was more "conditional" and less "temporal".

cbrnr, 2018-05-05:
Sure, we should check. What I meant to say though was that we have these special installs even for Python 2, so this is not a Python 3 issue. Of course we could check if the normal versions work in case no one has done that in a while. @larsoner @christianbrodbeck?

cbrnr, 2018-05-05:
The docs recommend using `pip` (via [environment.yml](https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml)) to install MNE and not conda-forge, and everything seems to work with Python 3.

cbrnr, 2018-06-01:
Now that mayavi is pip-installable on Python 3, I think we can safely implement the Python 3 transition following the example of NumPy. I can write a short text, but where should this text go? It could be a short section in README.rst and/or a short section in the documentation section of the website (e.g. in "Getting started" right after "Introduction to Python"). Let me know what you prefer. I will then also try to add MNE to https://python3statement.org/.

larsoner, 2018-06-01:
I would do the README plus `whats_new.rst` API section, plus maybe a `.. note` in `install_mne_python.rst`. That way the plan will be made clear to people when we release 0.17. It's a bit redundant but we do want to warn people they need to transition before 0.18.

PR #5256 (2018-06-04): [MRG+1] Add notes for Python 3 transition
Fixes #5167.
closed 2018-06-04 by larsoner