mne-tools/mne-python #5168: documentation and example inconstancy

massich, 2018-04-26:
https://github.com/mne-tools/mne-python/blob/057026f9a38c63607401ca568790202cf57a08c8/mne/commands/mne_flash_bem.py#L36

According to xxx the call should either be 
```
$ mne flash_bem --subject=sample
```
or 
```
$ mne flash_bem --s sample
```

larsoner, 2018-04-26:
> or
> $ mne flash_bem --s sample

I think this should be `mne flash_bem -s sample`. The Unix standard is single dashes with single letters, double dashes with longer ones. Which is consistent with the docstring:
```
  -s SUBJECT, --subject=SUBJECT
                        Subject name
```

larsoner, 2018-04-26:
Ahh I see now, you mean in particular the `=` is missing. (My visual `diff` missed this difference.) +1 for updating it, even though in practice I don't think it matters. For example this works:
```
mne browse_raw --raw ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif
```
This is either a bug or feature in `optparse` (I'm assuming a feature).

massich, 2018-04-27:
Yes I did a typo while writing the issue. And its either we make it short or we add the `=`, for the example.

not using the `=` was breaking for me.

agramfort, 2018-08-04:
@massich can we close or anything to fix?

PR #5414 (2018-08-06): [MRG][DOC] fix mne_flash_bem ussage 
fix #5168
commit c78271f9 (2018-08-06): [skip ci][skip av][skip travis] fix #5168 (#5414)
closed 2018-08-06 by agramfort