mne-tools/mne-python #5171: ValueError in MVPA tutorial

katarinaslama, 2018-04-29:
Hi,

I'm trying to run this tutorial using mne 0.15.2: [https://www.martinos.org/mne/stable/auto_tutorials/plot_sensors_decoding.html]. 

When running coef = get_coef(time_decod, 'patterns_', inverse_transform=True), I get the following error:

'ValueError: This estimator does not have a patterns attribute.'

If I simply leave out 'patterns_' to let it default to 'filters_', by running the following line:
coef = get_coef(time_decod, inverse_transform=True), I get an equivalent error:

'ValueError: This estimator does not have a filters_ attribute.'

Can you help? Thank you very much for your time!

Best wishes,
Kata

dengemann, 2018-04-29:
Hi Kata,

Did you use the LinearModel wrapper class as demonstrated here
http://martinos.org/mne/stable/auto_tutorials/plot_sensors_decoding.html?highlight=linearmodel
?

Best,
Denis

katarinaslama, 2018-04-29:
No, I didn't! That fixed it. Thank you so much for your quick response.

closed 2018-04-29 by katarinaslama