mne-tools/mne-python #5172: BUG? flake8 fails locally but succeeds on travis

mathurinm, 2018-04-30:
I was happy to read in [https://mne-tools.github.io/stable/contributing.htm](https://mne-tools.github.io/stable/contributing.htm) that I can run `make flake` from the root to detect flake8 violations, instead of letting a Travis build run for 30 minutes before failing.

However I get lots of errors locally (lots of D400 First line should end with a period, D205 1 blank line required between summary line and description) while the Travis build succeeds.

It's a problem for me because it makes it difficult to check for non-tolerated pep8 violations locally. 

Am I missing something?

larsoner, 2018-05-01:
Did you `pip` install what is the version? Sometimes `flake8` updates and we have to either fix the changes or add them to the ignore list in `setup.cfg` (depending on difficulty and group preference).

mathurinm, 2018-05-01:
I had a pip installation indeed, I removed flake8 with pip and reinstalled it via conda, but it still failed. 
Inspecting with `conda list` gave me:
```
mathurin@mathurin-HP-EliteBook-840-G3:~/workspace/mne-python$ conda list flake8
# packages in environment at /home/mathurin/anaconda3:
#
# Name                    Version                   Build  Channel
flake8                    3.5.0                    py35_0    conda-forge
flake8-docstrings         1.0.3                     <pip>
flake8-polyfill           1.0.1                     <pip>
```

and removing the last 2 packages (with pip uninstall) makes it work. Not sure why.

closed 2018-05-01 by mathurinm

larsoner, 2018-05-01:
Flake8 must have optional components. We use pydocstyle currently for
docstring checks. You could make flake8 check the same set if you want.

>