mne-tools/mne-python #5176: MNE Coregistration GUI Black Viewing window on MacOsX

tbardouille, 2018-05-01:
Hi there,
I'm using MNE python 0.15.2 on a Mac (OsX v.10.13.1). When I run mne.gui.coregistration(), the GUI opens, but the viewing window (where the MRI would appear) is black.
At the same time, I get a number of messages to standard output (see below).
Any ideas as to what I'm doing wrong?
Thanks,
Tim.

Standard out messages:
TAFC5:~ tbardouille$ source activate mne
(mne) TAFC5:~ tbardouille$ ipython
Python 3.6.5 |Anaconda, Inc.| (default, Apr 26 2018, 08:42:37) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.3.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import mne

In [2]: mne.gui.coregistration()
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:363: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  version=int(np.fromstring(fid.read(4), dtype=">i4")),
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:364: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  machid=np.fromstring(fid.read(8), dtype=">i4"),
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:365: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  secs=int(np.fromstring(fid.read(4), dtype=">i4")),
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:366: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  usecs=int(np.fromstring(fid.read(4), dtype=">i4")))
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:244: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  ndim = int(np.fromstring(fid.read(4), dtype='>i4'))
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:246: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  dims = np.fromstring(fid.read(4 * ndim), dtype='>i4')[::-1]
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:263: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(read_big(fid, 4 * dims.prod()), dtype='>f4')
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:244: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  ndim = int(np.fromstring(fid.read(4), dtype='>i4'))
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:246: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  dims = np.fromstring(fid.read(4 * ndim), dtype='>i4')[::-1]
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:263: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(read_big(fid, 4 * dims.prod()), dtype='>f4')
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:244: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  ndim = int(np.fromstring(fid.read(4), dtype='>i4'))
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:246: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  dims = np.fromstring(fid.read(4 * ndim), dtype='>i4')[::-1]
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:259: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  data = np.fromstring(read_big(fid, 4 * dims.prod()), dtype='>i4')
Using normal I/O
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
    Creating tag directory for /Users/tbardouille/Documents/Work/Projects/CambridgeLargeData/Data/release004/subjects/fsaverage/bem/fsaverage-fiducials.fif...
start { 0
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:363: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  version=int(np.fromstring(fid.read(4), dtype=">i4")),
[... 12 lines cut ...]
    end } 107
    block = 0 nent = 4 nchild = 1
end } 0
[done]
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:178: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  out = np.fromstring(fid.read(tag_size), dtype=dtype)
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:372: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  kind=int(np.fromstring(fid.read(4), dtype=">i4")),
[... cut ...]
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:373: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  ident=int(np.fromstring(fid.read(4), dtype=">i4")),
/Users/tbardouille/anaconda3/envs/mne/lib/python3.6/site-packages/mne/io/tag.py:374: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
  r=np.fromstring(fid.read(12), dtype=">f4"),
Out[2]: <mne.gui._coreg_gui.CoregFrame at 0x82930e4c0>

larsoner, 2018-05-01:
We are in the process (today) of releasing 0.16, which contains many fixes, including an updated `environment.yml`. Can you try the instructions here to update your environment:

https://mne-tools.github.io/dev/install_mne_python.html

Then install `mne` from GitHub, or wait until the end of the day for a release and just do `pip install --upgrade mne`.

Since you are on OSX you might need to do `pip install --upgrade pyqt5>=5.10` after the `source activate mne` step (the instructions will probably be updated soon to note this).

tbardouille, 2018-05-01:
To install from Github, do I just run the steup.py file in the distro?

https://mne-tools.github.io/dev/install_mne_python.html

larsoner, 2018-05-01:
You can do `python setup.py install` if you want a static one, or `python setup.py develop` if you want changes in your checked-out version to take effect as you change or update branches.

nugenta, 2018-05-02:
Eric, we spoke yesterday about what was essentially the same problem.  And I fixed with essentially the same solution, which was to use pip to update pyqt to pyqt 5.10.  However, I realized today that when you use pip to update pyqt, you end up with two versions of pyqt5 - conda calls it pyqt, while pip calls it PyQt5.  

https://github.com/ContinuumIO/anaconda-issues/issues/1554

Spyder, Jupiter, and qtconsole depend on pyqt, even after PyQt5 is installed. mne coreg works fine, but Spyder throws errors (although it seems to work).  Jupyter opens without errors, but I didn't test further. 

The solution I found is to remove spyder, jupyter, and pyqt from the conda dependancies, add jupyter, pyqt5, and qtconsole to the pip: section. 

In addition, you need cython in the dependencies, or you get errors for mkl-random and mkl-fft. 

There's no easy solution for spyder, because the conda version of spyder relies on pyqt. 
The version of spyder on pip requires pyqt<5.10

larsoner, 2018-05-02:
@nugenta I tried the instructions here yesterday on macOS High Sierra and it seemed to work (from #5174):

https://7923-1301584-gh.circle-artifacts.com/0/html/install_mne_python.html

I don't get errors when loading Spyder.

Can you try creating an env from scratch again (you can even use a different name) just to confirm?

tbardouille, 2018-05-14:
Hi Eric,

I still had problems with the new version of MNE-python. However, I made a second user on my OsX and the co-registration GUI works fine for that user. Clearly, there is a conflict with some other package installed on my main user account.

Although this doesn’t get to the root of the problem, it is a sufficient fix for me.

Thanks,
Tim.

https://7923-1301584-gh.circle-artifacts.com/0/html/install_mne_python.html

larsoner, 2018-05-14:
@tbardouille good to hear it works for the new user. In theory if you uninstall and reinstall Anaconda for your main user (the same way you installed it for your new one) everything will work. If this is too drastic, you can also just remove the `mne` environment like:
```
conda remove -n mne
conda create -f environment.yml
...
pip install --upgrade pyqt>=5.10
```

closed 2018-05-14 by larsoner