mne-tools/mne-python #5180: add support for xdf (labstreaminglayer format)

dmedine, 2018-05-02:
I think this should be added. I am happy to do this. There is already a python loader for the xdf format so it is just a matter of making output structure compliant with mne.

dmedine, 2018-05-02:
> would be great

Ok. I'll put this on my todo list and submit a PR when it is done.

sappelhoff, 2018-05-03:
That's a good idea. In the repo for the python XDF loader, there is an ongoing discussion to make a "pyxdf" package, however the discussion is stagnating a bit. Mainly because there is a lack of tests for the robustness/reliability of the loader. See [here](https://github.com/sccn/xdf/pull/10) for the discussion.

Integrating the xdf loader into MNE would not circumvent the issue of (so far) lacking tests, but maybe the stagnation is over now :-)

cbrnr, 2018-05-04:
@sappelhoff I think MNE would be a great place for the XDF loader!

larsoner, 2018-05-30:
I thought that we discussed this somewhere, but a low-order spline interp (or even linear!) would probably be sufficient. So I don't see this as too big a hurdle.

cbrnr, 2018-05-30:
Yes, it shouldn't be too difficult, just wanted to bring this up in case this was not obvious. Can you confirm that all signals must have the same sampling frequency? If so, how do we choose which sampling rate to resample to? Most XDF files have nominal sampling rates for each stream, so we could resample to the highest.

larsoner, 2018-05-30:
> Can you confirm that all signals must have the same sampling frequency?

Yes

> If so, how do we choose which sampling rate to resample to? 

Resampling to the highest data channel rate (by default, we could make different options with ``resample='auto'`` kwarg) makes sense to me. In practice I imagine these will all usually match anyway.

sappelhoff, 2018-08-09:
There are some developments with the python reader for XDF files. Soon, there might be a pip installable package, see: https://github.com/sccn/xdf/pull/21

That would make this issue easier to solve.

trmullen, 2018-11-13:
this is a great idea, and we would be happy to support this initiative!

cbrnr, 2018-11-14:
@dmedine @trmullen do you have time to work on this?

sappelhoff, 2018-11-14:
@cbrnr check out the discussion in the XDF issue linked above, specifically there is probably the need for some MNE decision on this statement from @dmedine:

> I said a long time ago that I would shove xdf into MNE. The issue there (if I understand correctly) is that if there are multiple streams, they all have to have the same sampling rate, first time-stamp, etc. This makes it labor intensive enough for me to avoid doing it. **However, if MNE is like EEGLab and it is content to grab only EEG and marker streams, then this might not be such a chore after all.**

@larsoner @agramfort 

Personally, I think that being able to extract an EEG stream from XDF is better than nothing and worth an implementation. An even better solution would be to allow the extraction of *any* single stream out of the XDF file.

... the "best solution" of reading the complete XDF file with all its streams seems tricky indeed

cbrnr, 2018-11-14:
I thought that the existing XDF reader already supports synchronizing time stamps from different streams, so this is one problem already solved. Grabbing a single stream is straightforward, because all streams are returned in a list of dicts, of which we'd need the values for the `'time_series'` and `'time_stamps'` keys.

agramfort, 2018-11-14:
MNE only supports "streams" with the same sampling rates. So yes if we can
read the EEG data and mimic what we have already for edf files +1

I also feel that we need more integrated efforts between pyedf, pybv,
pyxdf, neo and MNE to avoid duplication of efforts. I am sure I have time
though to coordinate these efforts. It would be a great topic for a code
sprint.

cbrnr, 2018-11-14:
Yes, absolutely! I also have the feeling that we could greatly benefit from closer collaboration. I'm already involved in MNE and pyxdf, so I could at least try to coordinate a bit.

dmedine, 2018-11-14:
Just to clarify, the XDF reader does synchronize the time stamps in that 
it puts all the streams on the same timebase, but it does not resample 
the data. So time 3.14 in stream A is the same as 3.14 in stream B, but 
they might have completely different sampling rates. Is this also an 
issue for markers, by the way?

dmedine, 2018-11-14:
I currently don't have time to work on this, but I can probably devote 
some in the future.

cbrnr, 2018-11-14:
Resampling to a common sampling rate shouldn't be a big issue as long as we start from tmin and include all time points up to tmax.

agramfort, 2018-11-14:
for my curiosity how do you remap to the same time stamps? do you
interpolate? how do you decide the regular grid of time samples?

cbrnr, 2018-11-14:
Yes, I'd resample all signals to the highest sampling rate, this will create samples at the same time points. In SigViewer, we do that, but the user can change the resampling rate if the one we choose automatically is too high. If the starting points differ, we add zeros to the signals starting later.

dmedine, 2018-11-14:
The way LSL puts everything on the same time base is fairly complicated. 
I will try to summarize it in broad strokes.

Basically, when data is being streamed by an outlet into an inlet, there 
are periodic handshakes between inlet and outlet that return the time 
difference between the CPU clock times between the two. The algorithm 
for doing this is canonical and based on the NTP method for doing the 
same. In the LSL recording program, these offsets are written into the 
footer of the data file as the streams are being recorded. Meanwhile 
each datum is also timestamped based on its own CPU time. Then, when the 
file is loaded, the loading program (implemented in Matlab and Python) 
uses this information to form a piecewise linear map between the 
timestamps in each incoming stream and the write times in the recording. 
This corrects the slowly drifting time offsets between various sending 
CPU times and the CPU time of the recording.

If the stream is coming in at a regularly sampled sampling rate (e.g. an 
EEG stream, not an arbitrary marker stream) the differences between 
successive timestamps are also smoothed to correct for jitter. The end 
result is a data structure that has the data for each stream and its 
corresponding, corrected, timestamp values. These streams are then said 
to be synchronized. This has been validated to provide sub-millisecond 
accuracy. This doesn't align the samples to a common grid. That would, 
of course, require an interpolation step, which is the part that I don't 
have time to work on right now. This would, now that I think about it, 
perhaps be a nice feature to add to the LSL loading software. So far as 
I know, no one has ever requested this.

What isn't corrected for (it must be measured) is the lag time between 
hardware acquisition and the time the data gets timestamped. There are 
various means of measuring these relative latencies, but it isn't always 
so straightforward. Back in the day this required a fair amount of 
handmade microphones, photo-diodes, and buckets of salt water. Brain 
Products (shameless plug) sells hardware (StimTrak 
<https://www.brainproducts.com/productdetails.php?id=57>) that can 
provide these measurements through sending hardware triggers. NBS also 
sells a product (which is LSL based) called the LabStreamer 
<https://www.neurobs.com/menu_order/purchase_labstreamer> that helps 
measure these hardware latencies.

One last detail is that LabRecorder (the LSL recording program) picks up 
each stream on a separate thread so they always have a different start time.

agramfort, 2018-11-14:
thanks a lot for clarifying

then for MNE what we need is a way to resample all streams we want to read
to the same sampling freq
like we do for EDF reader.

trmullen, 2018-11-16:
Hi @cbrnr -- yes happy to assist with this effort.

Tim

jasmainak, 2018-11-16:
Not adding much to the discussion. But cross linking to [related discussion](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/bids-discussion/NeqFO-lJhsc/_atcTFfIBQAJ) on BIDS mailing list

cboulay, 2018-12-12:
pyxdf is now available with `pip install pyxdf`

cbrnr, 2019-03-21:
As an initial implementation, what about allowing only one regularly sampled stream (plus optionally any number of marker streams)? That way, we don't have to resample and we can directly use time stamps and data samples provided by `load_xdf`. Optional marker streams could be converted to annotations. This would probably cover many use cases already, and would be a lot better than what we currently have (nothing).

agramfort, 2019-03-22:
> As an initial implementation, what about allowing only one regularly
> sampled stream (plus optionally any number of marker streams)? That way, we
> don't have to resample and we can directly use time [...]
no objection but I would start with a gist

cbrnr, 2019-03-26:
Here's a first proof-of-concept implementation: https://gist.github.com/cbrnr/7164a12868bc7616056587a41f51e92d

Basically, it loads the first EEG stream it finds and adds the first marker stream if present. For many XDF files, https://github.com/sccn/xdf/pull/42 should be fixed before this works. Comments and suggestions welcome!

agramfort, 2019-03-26:
would this work with preload=False?

>

cbrnr, 2019-03-26:
Because XDF is a format designed for streamed data, `preload=False` would require to parse the whole file. This is certainly doable, but I think it should be implemented upstream. What other requirements does MNE impose on readers? Blockwise random access reading?

larsoner, 2019-03-26:
> What other requirements does MNE impose on readers? Blockwise random access reading?

In general yes, one should at a minimum (that I can think of):

1. Populate `info` properly,
2. Give the number of samples in the file, and
3. Provide a function that returns an arbitrary chunk of data for a given `start, stop` sample number pair

If these are provided, all methods of `BaseRaw` should work (I think!).

If they are not / cannot be provided, we'd have to think about how to deal with this.

cbrnr, 2019-03-27:
I think all of these points can be implemented, but only with `preload=True`. XDF makes it really hard to randomly access a portion of data, because it does not reserve a constant amount of bytes per data segment. In any case, jumping around the file is still reasonably fast using boundary chunks, but since these are optional the worst case scenario is that the file must be searched from beginning to end. All of this should be provided upstream, but currently `load_xdf` always loads everything.

agramfort, 2019-03-27:
pyedf or pyxdf don't support random access of chunks at all? I don't have a
strong feeling on this but that would make a precedent to have readers that
don't support preload.

>

cbrnr, 2019-03-27:
`pyedflib` does support random access, that is you can specify the channel, start time, and end time. I think XDF is really an exception here - I don't know any other file format that's structured similarly, so this might be the only reader for now.

cbrnr, 2019-03-27:
I've added a `parse_xdf` function that does a quick pass over the whole file to read just the headers (but ignore all actual content). This is very fast even for large files (fractions of a second) as compared to the complete `load_xdf` call, which can take several seconds. Maybe we can use `parse_xdf` to fill our meta info even in the case of `preload=False`. But once data is actually needed, we'd need to pull in everything, so it's just a simple lazy loading without random access.

agramfort, 2019-03-27:
not strong feeling here. Do you need this to process your own data? or some
public data?

>

cbrnr, 2019-03-27:
I don't even work with XDF myself, but I have been involved with the format and I also needed it in SigViewer. But I'm not sure if this was your question...

tstenner, 2019-03-27:
> In general yes, one should at a minimum (that I can think of):
>
> 1.    Populate info properly,
> 2.  Give the number of samples in the file, and
> 3.    Provide a function that returns an arbitrary chunk of data for a given start, stop sample number pair

All of these require at least one full pass over the file because the number of streams and samples isn't known in advance. Also, with postprocessing the number of samples might change, even for evenly sampled streams.

I'm not an expert, but I think the postprocessing in the current implementation needs all samples and can't just take a time slice, select all chunks which fall into that range and then read only those samples. Plus, the default of a few seconds of data per chunk means that most operations will need to read multiple chunks so preloading might make everything a lot simpler and faster, even if only a subset of data is needed.

cbrnr, 2019-03-27:
> All of these require at least one full pass over the file because the number of streams and samples isn't known in advance. Also, with postprocessing the number of samples might change, even for evenly sampled streams.

Yes, but the pass is much quicker with `parse_xdf` (i.e. when you don't process the contents). Maybe we can use this to fill our meta info when `preload=False`.

> I'm not an expert, but I think the postprocessing in the current implementation needs all samples and can't just take a time slice, select all chunks which fall into that range and then read only [...]

Correct, once you start working with the actual data, everything needs to be read and processed (clock synchronization). So basically random access just won't be possible.

cbrnr, 2019-03-27:
BTW, `read_raw_xdf` in https://gist.github.com/cbrnr/7164a12868bc7616056587a41f51e92d also supports loading arbitrary streams by name (and since names need not be unique I could/should also add loading by stream ID). If no name is provided, the first EEG stream is used.

larsoner, 2019-03-27:
I'm okay with only allowing preload=True or string for this format then

massich, 2019-03-27:
in `0.17.1` this is already the case that we can only read with `preload=True`. i.e: edf files with annotations.

agramfort, 2019-03-27:
ok fair enough then. I assume we can use as soft dep
https://pypi.org/project/pyxdf/ ?

cbrnr, 2019-03-28:
We could, but this should be discussed again. It looks like MNE is much more actively maintained than pyxdf, so we could also think about integrating the XDF code directly into MNE. I'd prefer this.

agramfort, 2019-03-28:
I would prefer to boost the maintenance team of pyxdf and eventually pyedf
to distrbute the maintenance load. It would scale if we always integrate
the community work.

>

sappelhoff, 2019-03-28:
Couldn't you get write permissions for pyxdf @cbrnr ? That would at least remove potential "blocks" due to inactivity from the reviewing / merging side of PRs at pyxdf?

cbrnr, 2019-03-28:
I'd hope that I could (@cboulay WDYT?), but I can't just rage-merge everything without review 😛 - but it would certainly help if I was more directly involved.

BTW, is there a way I can list all GitHub repos I have write access to?

cboulay, 2019-03-28:
I've been explicitly told that I cannot add any new people to SCCN repos. I recently asked for help and guidance, because it's clear that xdf requires more stewardship than it currently receives (and I don't have time to give), and I received a couple "Chad's right, we need more work on this" and a couple no-answers.

One of my ideas, which was not well received by the xdf oversight committee (*real people but not a real committee), is to break out xdf into submodules just how we did it for labstreaminglayer. Then we can have individuals working with more authority on the code while not granting them the ability to break SCCN's repositories. i.e., any changes that happen upstream in these submodules will not be visible to sccn/xdf users until someone with SCCN authority updates the submodule pointers.

Even though there was resistance to this, I went ahead and created the xdf-modules GitHub organization and put in a few placeholders for submodules it will hold. I'm just asking around now about how to move the code without losing the commit history.

I've already invited a few people to the organization, including @cbrnr , but there isn't much to be done yet until we get the code + commit history moved over.

agramfort, 2019-03-28:
let me share my thoughts on this. If you want xdf to become a community
standard you need to make pyxdf a project that is not affiliated to SCCN.

if you can move the code to a more neutral org it's great. thanks a lot

>

cbrnr, 2019-03-28:
I think creating separate repos is a step in the right direction (thanks @cboulay for adding me) - I always thought that `pyxdf` shouldn't be combined with MATLAB (and potentially other stuff). I agree with @agramfort, but that's exactly what @cboulay just did - if the new upstream will be xdf-modules/pyxdf, then it is essentially detached from SCCN. In any case, since at least the pyxdf code is BSD licensed, we can always fork it and roll our own thing (that's why I was suggesting to do it in MNE, it's really not that much more to maintain).

agramfort, 2019-03-28:
forking and maintaining our own code is bad community-wise. It make the
community more fragmented while we should consolidate

>

cbrnr, 2019-03-28:
Yes, I meant migrating and not forking.

cboulay, 2019-03-28:
@cbrnr you now have write access to https://github.com/xdf-modules/xdf-Python
I will ask the other people who have made Python pull requests to move their PRs to that repo instead.

cbrnr, 2019-03-28:
Thanks. BTW, I've never migrated/transferred a repo, but shouldn't it be possible with the standard git commands (i.e. cloning and pushing to the new repo)? https://www.atlassian.com/git/tutorials/git-move-repository

cbrnr, 2019-05-02:
I've implemented a basic XDF reader in MNELAB:
https://github.com/cbrnr/mnelab/pull/22

There are some caveats (see my comment in that PR), but loading basic EEG streams should work. I'm not sure if we can handle loading multiple streams, because everything gets very slow even with just one long stream (with a high number of channels/samples, e.g. 80 channels, fs 1000Hz, 10 minutes).

teonbrooks, 2019-05-02:
@cbrnr very cool. it looks like we might be able to incorporate the lazy evaluation of preload and seek on demand. with xdf, can streams be added during recording or we do always know that it would contain certain streams and only those stream throughout?

tstenner, 2019-05-02:
@teonbrooks New streams can appear anywhere in the file, but per spec the header for the stream has to appear before the data chunks.

teonbrooks, 2019-05-02:
I guess then you would need to be able to read all the headers first to and construct a map of pointers (sample_n --> data_block_n) for the eeg if you wanted to avoid loading the whole file

cbrnr, 2019-05-03:
Yes, it is possible to parse the whole file and just consider the data of the stream you want (and skip over the data you don't want), because each stream has a unique stream ID. That's something I'm already doing with `parse_xdf`, but the current version of `pyxdf` (which I use to actually load data) doesn't support this yet.

cbrnr, 2019-05-23:
@agramfort just thought about your comment - when you say pyxdf should not apparently be associated with SCCN, why is this not a problem with MNE Python hosted at Martinos?

agramfort, 2019-05-23:
>
> y@agramfort <https://github.com/agramfort> just thought about your
> comment - when you say pyxdf should not apparently be associated with SCCN,
> why is this not a problem with MNE Python hosted at Martinos?
>

mne-python is not hosted on an github org of martinos and it's not
administrated by anyone at martinos
we have the website historically hosted at martinos.org. We use
mne-tools.github.io as nightly build of the doc.
It's true that we have martinos in the url of the website but we have all
the logos of all the institutions
that helped over the years.

Ardillen66, 2020-12-08:
Hello, 

I would like to know what the status is on this feature in 2020, as it is still open. Are you still planning to include this at some point? I would be very interested in having this feature built into MNE.

Alternatively, it would be nice if you include a link to relevant software in the documentation if you are not planning to include this in MNE. That way people looking for a solution don't have to go to google and read through this issue.

[14 more replies]

PR #9289 (2021-04-20): Add XDF to "Reading EEG data" tutorial
Closes #5180.

Besides tidying up and improving the docstring a bit, I have added a section on how to import XDF data.
closed 2021-04-20 by larsoner