mne-tools/mne-python #5181: Digitalization source is not flush when changing subject in MNE Coreg

massich, 2018-05-02:
When processing several subjects, when selecting a different subject the points on the display are flushed, but not the digitalization file. This could lead to problems like fitting the wrong digitalization. 

See the following images:
[image]
[image]

larsoner, 2018-05-14:
I'm not sure that the dig file should change when loading a different MRI. When using a surrogate / warping someone might want to choose MRI based on closest fit to the dig, in which case clearing the dig when the MRI changes would be very annoying.

agramfort, 2018-05-14:
well for me if you change the subject you have a new MRI and you need to
update dig points.

it's less surprising. We got bitten by this by @massich and this triggered
this issue.

christianbrodbeck, 2018-05-14:
Maybe the logic of the GUI, assuming that people work from top to bottom, would be more intuitive if the digitizer source was above the MRI source?

agramfort, 2018-05-14:
for me this does not fix the pb. if you switch subject in your study you
want to flush what is not relevant

larsoner, 2018-05-14:
> well for me if you change the subject you have a new MRI and you need to update dig points.

Sure, but this is just one of many possible workflows. In this workflow, I agree in this case it could be less surprising to remove it.

However it is surprising in other use reasonable (and not uncommon) use cases:

- in the case where a user has their `subjects_dir` selected (likely the default), then picks their dig source, then picks the MRI subject, what you would propose would remove their dig source.
- in the case where someone is looking for a surrogate MRI that fits a dig source

Moreover, in `mne_analyze` you can change MRI subject and the dig info is never removed.

So I'm still -1 on making an MRI change remove the dig source by default. Maybe a (off by default) command line option `--clear-dig` that removes the `dig` if the `subject` is changed?

agramfort, 2018-05-14:
ok I see the use case and the rationale then. I will be able to argue like you next time
I hear someone complain :)

I guess we can close

closed 2018-05-14 by agramfort

larsoner, 2018-05-14:
Yes we can classify this as a [BEBKAC](https://en.wikipedia.org/wiki/User_error) bug :)