mne-tools/mne-python #5190: plotting: topomap electrode positions change depending on number of channels

sappelhoff, 2018-05-06:
When using topoplot functions in MNE, I run into problems with the electrode positions that are plotted. Namely, when using the code below, the resulting sensor plots should look identical for the electrodes in `chan_selection`. However, this is not the case:

```python
import mne

chan_selection = [u'FC5', u'FC1', u'C3', u'CP5', u'CP1', u'FC2', u'FC6', u'C4', u'CP2', u'CP6']

montage1020_full = mne.channels.read_montage(kind='standard_1020', ch_names=None)
montage1020_reduced = mne.channels.read_montage(kind='standard_1020', ch_names=chan_selection)

montage1020_full.plot()
montage1020_reduced.plot()

```

## Full
[image]

## Reduced
[image]

---------------------------

I am aware of a bunch of related issues (e.g., #4880). Issue #3987 seems to be the central issue for this problem ... is anyone working on this currently? It seems like a pretty crucial problem to be solved, given that topoplots are a very common visualization in our domain.

Is this perhaps even affecting the interpolation of voltages across the scalp? Or is this problem merely distorting the "plotted dots" symbolizing the electrodes?

PS: When I use real data and `mne.viz.plot_evoked_topomap`, I get yet another image for the same electrodes:

[image]

larsoner, 2018-05-06:
#5085

sappelhoff, 2018-05-06:
Mmmh okay, those are a lot of scattered issues :-) but it seems like you are working on it.

Especially this point (quoting @larsoner)
> This will fix issues like evoked.plot_topo() and evoked.pick_channels(evoked.ch_names(:10)) not having the 10 sensors in the same position (which is a bit crazy IMO).

For now, is there a workaround, I could use? I tried supplying a `head_pos = dict(center=(0., 0.), scale=(1., 1.))` with different scales, AND in conjunction with specifying a layout derived from an info object. However, the results are not really satisfactory and it feels like fishing around ... 

On this note ... what is the difference between [layout](https://martinos.org/mne/dev/generated/mne.channels.Layout.html) and [montage](https://martinos.org/mne/dev/generated/mne.channels.Montage.html?highlight=montage)?

agramfort, 2018-05-06:
I agree it's quite a nasty bug...

larsoner, 2018-05-06:
> but it seems like you are working on it.

Yes passively currently but hopefully actively again soon.

> For now, is there a workaround, I could use?

You have the right idea, but it won't work properly yet because the interpolation is still done over the entire head (kind of nonsensically).

> On this note ... what is the difference between layout and montage?

I have updated my plan for this here:

https://github.com/mne-tools/mne-python/issues/3987#issuecomment-298920228

Basically `montage` / `DigMontage` / physical locations are to be used anytime sensors are shown relative to a head. `layout` is to be used to separate sensors to facilitate subselection by groups e.g. in `raw.plot`. But this is still TODO, we are not consistent with this yet.

PR #7066 (2019-12-11): MRG, BUG: Fix M/EEG topomap plotting
Closes #3987
Closes #4880
Closes #5190 
Closes #5471
Closes #5472 
Closes #6304

Todo:
- [x] ensure our montages actually use the same head sizes (done by dig refactoring)
- [x] Make the `plot_sensors` be in physical coordinates
- [x] make all code assume a 0.095 m head radius
- [x] check that all examples in gist are fixed
- [x] deprecate params: `transform` in `montage`, `outlines`, ...
- [x] get tests to pass again
- [x] change to `sphere` rather than `head_radius` for consistency and flexibility
- [x] change units in `setup_volume_source_space` for cons [... cut]
closed 2019-12-11 by agramfort