mne-tools/mne-python #5191: Saving RtEpochs

hekolk, 2018-05-07:
TL;DR RtEpochs do not support all of the inherited ``BaseEpochs`` methods. It would be nice to have an efficient conversion-method to an ``EpochsArray``.

While ``RtEpochs`` are useful for gathering and processing individual  epochs, it would be helpful to be able to make them completely compatible with an "offline" ``Epochs`` object. For example,   currently it is not possible to save a ``RtEpochs`` object due to  (at least) the following issues:
     - missing parameters (``out`` and ``verbose``) in ``_get_data``
     - ``drop_log`` is missing or incorrect
     
As a workaround, I exported the data to an EpochsArray [1], but noticed that data needs to be copied multiple times:
  1. a single numpy array into which all ``_epoch_queue`` items are copied
  2. the Epochs constructor again copies the data in its call to ``decimate`` (which does not make sense to me anyway, since I would expect a ``decimate(1)`` call to not have any side-effects)

Having three (temporary) copies of the epochs of a full experiment in memory leads frequently to out-of-memory situations on typical workstations.

I would propose to 
  - remove the mandatory call to ``decimate`` or omit the copying when creating data from an existing array (although there is probably a reason the call is currently implemented?)
  - either ensure that ``RtEpochs`` *is a* ``BaseEpochs`` object supporting all the methods or alternatively provide a conversion method to an ``EpochsArray`` etc.

It would be great to hear your thoughts on this. I'm willing to contribute.

larsoner, 2018-05-07:
I think making `RtEpochs` behave like regular `Epochs` is the best way to go, it just hasn't seen as much use so probably doesn't fully comply currently.

- I'm okay with having `decimate(1)` short-circuit without making a copy as long as we document it in the function (and note this in the API section of `whats_new.rst`).
- Fixing the `drop_log` (and presumably also `selection`) should be doable. There are `reject` and `flat` parameters so some epochs can be dropped, so the `drop_log` is useful. We should always have these relationships:
  ```
  assert len(epochs.selection) == len(epochs.events) == sum(len(dl) == 0 for dl in epochs.drop_log)
  assert len(epochs.drop_log) >= len(epochs.events)
  ```
  We should probably add these to the `epochs` constructor somewhere. Maybe a `BaseEpochs._check_consistency()` could do these things, and be called in `BaseEpochs.__init__()` and `BaseEpochs.save`.
- Fixing the `_get_data` should also be doable.

jasmainak, 2018-05-07:
+1 for fixes to `RtEpochs` and would love to see a PR.

hekolk, 2018-05-08:
I addressed ensuring the invariants and fixing ``RtEpochs`` in #5193 . 
The short-circuiting of ``decimate(1)``, however, doesn't work out-of-the-box since it is seems to be needed to get single-segment arrays (for which ``data.resize`` works) - I'll leave that as-is for now.

larsoner, 2018-05-08:
This seems like a `numpy` problem we can solve with a `np.ascontiguousarray` or `np.require` or some similar call, if you are motivated to look into it. Otherwise I'm fine with leaving it

PR #5193 (2018-06-12): MRG: Fix RtEpochs to support saving
Fixes #5191: Make sure the epochs invariants hold  for ``RtEpochs``.

I'm not completely happy with the ``_get_data()`` method since it relies on the "online iteration and blocking" over epochs rather than just returning the data received so far (which I would prefer, but didn't implement for compability).  I had to move the check for the timeout since otherwise iteration is typically only possible once (depending on the ``isi_max``).
The inherited implementation of _get_item was broken (but is called by ``save``).
closed 2018-06-12 by larsoner