mne-tools/mne-python #5192: eeg ica without eog

pdeman, 2018-05-08:
Hi,

I am trying to do the solution presented [here](https://martinos.org/mne/stable/auto_tutorials/plot_artifacts_correction_ica.html):
What if we don’t have an EOG channel? methode 2
but after the first line of code it says:
reference_ica.plot_sources(eog_average, exclude=eog_inds)

and I don't have eog_average has I don't have eog.
is it a template to download somewhere ?

Regards,
Pierre

agramfort, 2018-05-09:
you're going to have to mark manually the EOG components

cf. exclude attribute of ICA object.

pdeman, 2018-05-09:
so I do:
reference_ica.plot_components()
I look wich one are suspicious on the ICA.
let's say number 2 as in the example.
I want to look at it in the time series to be sure, how I do that ?
and then I will have to do:
ica.exclude.extend([2]) ?

and then how do I do the ICA backprojection to get the denoised signal ?

agramfort, 2018-05-09:
@pdeman

I think this is documented but maybe not well enough...

you can script it with:

ica.exclude.extend(2)
or
ica.exclude += [2]

or just click on the components names that becomes grey with excluded
in ica.plot_components()

they see ica.plot_sources and ica.apply methods

cbrnr, 2018-05-09:
I've written a short summary on how to remove eye activity with ICA, maybe this is helpful for you: https://cbrnr.github.io/2018/01/29/removing-eog-ica/

closed 2018-05-12 by pdeman