mne-tools/mne-python #5195: beamformer group level (GSoC)

TommyClausner, 2018-05-09:
Hey there MNE community!

For the next ~3 months I'm going to work on implementing some necessary functions for performing group statistics on volumetric (beamformer) data. This will be my Google Summer of Code Project and in case you're interested you could check out the progress here: [https://blogs.python-gsoc.org/tommy-clausner/](https://blogs.python-gsoc.org/tommy-clausner/)

In order to maximize the outcome it would be nice to involve a broad audience / the MNE community. So any broader suggestions or feedback, related to the project, could be given here and are very welcome, while I will open separate issues for separate steps as appropriate (if that's alright?).

The final goal of the project would be to implement

- warping of volumetric grid spaces to another
- making pseudo-individual anatomical MRIs based on head-shapes
- shape the respective output desirably (would be a major thing to be discussed)
- (get into the plotting)

in order to prepare single subject data for group level analyses / statistics.

My plan was to start with existing toolboxes first (MNE, FreeSurfer, FSL, ANTsPy) and see what I can exploit in which way. So any experience or hints are very welcome! 
Afterwards I'd like to write some alpha-stage functions to do the warping in the style of beamformerData.morph('MNI') to e.g. morph a grid space to MNI space. Finally the output would be shaped such that it incorporates the existing API. Bonus: if time allows I'd like to get into the respective plotting functionality as well!

The next days I'll get a bit more into the MNE workflow / ecosystem and happily discuss any points that already might arise. 

A more detailed to-do list will follow soon, but feel free to drop any more general comment / suggestion / complaint / wish / question / etc  in here.

For now this issue serves as some kind of "Hello World" issue and project's wish list :)

Looking forward to working with you!

Cheers,

Tommy

larsoner, 2018-05-09:
> I will open separate issues for separate steps as appropriate (if that's alright?).

Agreed, this is where we can converge on API, etc.

> warping of volumetric grid spaces to another

When you say `warping` do you mean an affine transformation, or something else? Would this be based on coregistering the MRIs (based on brain shape)? I'm out of my depth on this a bit, but yes please start an issue to discuss this. It might help if you start by looking at how `setup_volume_source_space` is used in examples, and the MRI interpolator it already provides. This is what most of us will be familiar with.

> making pseudo-individual anatomical MRIs based on head-shapes

`mne coreg` already allows this to be done, feel free to try it and see if it's sufficient for your use cases. In the last few weeks we added transforming the MRI files (in addition to Freesurfer surfaces) by tweaking the `affine`.

We do have spherical warping implemented in principle, but in practice it would involve a bit of work (especially in terms of the MRI warping) to get it to work. You are welcome to tackle this if you are motivated to do it!

The current to-do list for `mne coreg` lives in #3934. So if it seems deficient feel free to comment there and once we converge I can add more to the top-level TODO list.

> shape the respective output desirably (would be a major thing to be discussed)

I have no idea what you mean :) Shape as in spatially deform, or shape as in `ndarray.shape`, shape as in "craft" / design an API, or ... ?

> (get into the plotting)

Feel free to open an issue for this, too. So far we have used `nilearn` plotting, and it would be great if we could continue to do that, or extend what they have available. If you want something to start with, take a look at #4496 (and feel free to take over if you have time!)

agramfort, 2018-05-09:
I would also list and experiment with  these options:

http://nipy.org/dipy/examples_built/affine_registration_3d.html
http://nipy.org/dipy/examples_built/syn_registration_3d.html

closed 2018-12-12 by larsoner