mne-tools/mne-python #5200: add_channels causes duplication in info['hpi_results'] and info['hpi_meas']

sarasommariva, 2018-05-11:
Hi, 

I think I found a small bug in the function `add_channels`.
When modifying an  existing data some of the fields (at least, `‘hpi_results’` and `‘hpi_meas’`) are duplicated causing some compatibility problems with e.g. neuromag.
Looking into the code, I think the problem is in the function `_merge_dict_values` . Both `‘hpi_results’` and `'hpi_meas’` consist of a list of dictionaries that seem to be concatenated by the function.

Here is a minimum example that demonstrates the issue:

https://gist.github.com/sarasommariva/d1c4b1a3015bf3ae33fefc4ffef2cbc9

agramfort, 2018-05-14:
I can confirm this unexpected behavior.

@larsoner you wrote this code I think. thoughts?

larsoner, 2018-05-14:
Yes these should not be duplicated, but rather checked to make sure they are consistent (assuming both instances have them, it's probably fine if one does and the other does not), throwing an error if they are not consistent.

agramfort, 2018-05-14:
+1

@sarasommariva can you give it a try?

larsoner, 2018-05-14:
That sounds about right. You can use `object_diff` before calling `_merge_dict_values` if you want

larsoner, 2021-02-19:
Looks like this was fixed by @sarasommariva in #5211 (thanks!)

closed 2021-02-19 by larsoner