mne-tools/mne-python #5204: mne.gui.coregistration does not override SUBJECTS_DIR

jasmainak, 2018-05-14:
It seems that selecting the `SUBJECTS_DIR` from the dropdown menu from `mne coreg` does not find the surfaces. However, manually setting it in the command line before starting `mne coreg` seems to work. This seems unexpected behavior to me. I get the following for the sample data:

[image]

Let me know if you need more information to reproduce.

larsoner, 2018-10-04:
Not a critical bug (and probably a pain to fix) so moving to 0.18

larsoner, 2019-09-11:
I can no longer reproduce, I'm assuming this was fixed with some of the coreg refactors

closed 2019-09-11 by larsoner