mne-tools/mne-python #5205: Amplitude of interpolated channels very small

ckiefer0, 2018-05-14:
Hello,

I encountered some weird behavior regarding the interpolation of bad channels using MNE.

In this [Dropbox folder](https://www.dropbox.com/sh/11rjc45bbg2b38m/AACUW6rtZKcJKitEMfbMIKVqa?dl=0) are two examples of data sets before interpolation and after interpolation (marked with 'bcc' in the filename). MEG channels 7, 10, 142, and 156 are dead in the original data sets. After interpolation, the channels still appear to be dead/ flat when plotting the data, however, looking at the raw data for these channels we can see that the values are now different from zero. Redoing the interpolation yields the same result for these data sets.

This is how you can reproduce my results:

    import mne
    raw_fname = '/path/to/file'
    raw = mne.io.Raw(raw_fname, preload=True)
    data = raw.get_data()

    # mark bads during plot
    raw.plot(block=True)
    raw.interpolate_bads(reset_bads=True)

    # plot result
    raw.plot(block=True)

    # compare raw data
    data_bcc = raw.get_data()
    plot data[6], data_bcc[6]

To test this further I selected three channels randomly (50, 126, 226) to be interpolated in addition to the dead channels.
For the empty room data set, channels 126 and 226 appear to be dead in the resulting plot. If I interpolate channel 50 I can clearly see variation in the plot, however, the amplitude seems to be only about half the size of the surrounding channels.
For the other data set all channels appear to be dead in the resulting plot.

If you need more information please let me know.

ckiefer0, 2018-05-14:
I talked to a colleague and he might have an idea what causes the problem with the interpolation. I will try to convey his thoughts.
`_interpolate_bads_meg()` calls `mapping = _map_meg_channels(info_from, info_to, mode=mode)` which implicitly uses  `_map_meg_channels(..., origin=(0., 0., 0.04))`

Changing the call to
`mapping = _map_meg_channels(info_from, info_to, mode=mode, origin=(0., 0., 0.01))`
has a fairly significant impact on the result.

The reason why this might help is because the default head-coordinate system used in Jülich is tilted by 45deg about the PA-axis.  point (0. ,0., 0.04) tends to be somewhere near the cortex.

larsoner, 2018-05-14:
We should probably expose this `origin` parameter to make it explicit that it matters

PR #5588 (2018-10-07): MRG: Expose origin parameter
Closes #5205.
closed 2018-10-07 by agramfort