mne-tools/mne-python #5207: Coregistration GUI - BEM only, no high-resolution scalp surface

tbardouille, 2018-05-14:
Hi there,
I'm having a problem with the coregistration GUI. When I load in a subject, I can only see them BEM and not the high-resolution scalp surface (see image below). 
Am I missing a file from my subjects_dir that the GUI is expecting?
Thanks,
Tim.

[image]

larsoner, 2018-05-14:
`coreg` looks for:
```
_high_res_head_fnames = (os.path.join(bem_dirname, '{subject}-head-dense.fif'),
                         os.path.join(surf_dirname, 'lh.seghead'),
                         os.path.join(surf_dirname, 'lh.smseghead'))
```
where `bem_dirname` is `SUBJECTS_DIR/SUBJECT/bem`. Do you have any of these files? In theory should be produced by a standard Freesurfer pipeline, but it's possible our list of files to look for is not complete.

Is your `-head.fif` dense? The reason we don't use this file is that sometimes it is dense and sometimes it is sparse (depending on how the user has things set up), so we can't really rely on it. If it's common enough that it is dense for people, we could try to load it and check if it's dense or sparse because they typically differ by at least an order of magnitude in vertex counts.

larsoner, 2018-05-14:
(and `surf_dirname` is `SUBJECTS_DIR/SUBJECT/surf`)

closed 2018-05-15 by tbardouille

larsoner, 2018-05-15:
Is there a good place we could have put this in the docs? Maybe the docs for `mne coreg` command line or `mne.gui.coregistration`?

larsoner, 2018-05-16:
@jasmainak it sounds like you are just starting to use `mne coreg` a bit, do you have time to make a PR with this information plus whatever other pitfalls you encounter? I'm thinking we can populate the `Notes` section of `mne.gui.coregistration`

jasmainak, 2018-05-17:
> @jasmainak it sounds like you are just starting to use mne coreg a bit, do you have time to make a PR with this information plus whatever other pitfalls you encounter? I'm thinking we can populate the Notes section of mne.gui.coregistration

Sure! But after 18th. I have a deadline tomorrow :)

larsoner, 2018-09-11:
I can't think of anything offhand, we can open new issues as people experience them (or @tbardouille if you still have problems let us know and we can reopen)