mne-tools/mne-python #5208: nonlinear warping source grid using nipy (GSoC)

TommyClausner, 2018-05-15:
Hey together,

regarding the volumetric transform, that is the core of my GSoC project:

I played around a bit with the example of [dipy](http://nipy.org/dipy/) to see, if it works in the way we want it.

And what shall I say? It does! Or at least it seems quite alright to me (see below):
[image]

What you see is: (top left) the subjects T1 overlaid with the beamformer result, (top right) fsaverage overlaid with the beamformer result, (**bottom left**) fsaverage overlaid with the **morphed** beamformer result and (bottom right) a checkerboard volume onto which I applied the same transform.

What I did was to use the lcvm [example of MNE](https://martinos.org/mne/dev/auto_examples/inverse/plot_lcmv_beamformer_volume.html) + the respective T1 + the fsaverage T1.

I first computed an affine transform like in the [dipy example](http://nipy.org/dipy/examples_built/affine_registration_3d.html), followed by the non-linear transform similar to another [dipy example ](http://nipy.org/dipy/examples_built/syn_registration_3d.html).

From the static volume (fsaverage) and the moving volume (subject's T1) only the brain as provided by FreeSurfer was used.

Afterwards I applied first the affine and then the nonlinear transform to the result of [apply_lcmv ](https://mne-tools.github.io/dev/generated/mne.beamformer.apply_lcmv.html) (saved as volume with mri_resolution=True). 

This mapped the source space result to fsaverge.

So: do you think that's alright?

Thanks a lot!

PS: Thanks @agramfort for the nipy / dipy hint ;)

agramfort, 2018-05-15:
great

can you share a script to replicate?

next is discussing API

is it worth comparing to other registration approaches?

TommyClausner, 2018-05-15:
Have to make the script readable first ;)
- shall I just post the code here or upload a file somewhere?

Was also wondering if it's worth looking further... Because actually I'm quite happy (except for the computation time)

agramfort, 2018-05-15:
thanks Jon for the enthusiasm

Tommy you could start to integrate this as an example.

By doing this you will face a number of issues:
- flake8 complaining
- need to update CIs setups (circle ci for example)
- other devs commenting also every line of your python code to suggest
better function/variables namings and function signatures.

larsoner, 2018-05-15:
Agreed turning this into a proper (advanced) example is a good way to go. After it works PRs can proceed by making this example simpler using APIs in `mne` namespace.

To update CIs you should be able to update `environment.yml` to include `dipy`. This should be picked up by CircleCI, which builds the example scripts. Eventually we can add a soft dependency on it and update README.rst, etc.

TommyClausner, 2018-05-16:
thanks @jhouck for the suggestion! Down sampling will surely help to reduce computation time - maybe it would be worth creating a variable for this. 

Thanks for being positive about it! I'm very happy to contribute to filling this gap ;)

@agramfort and @larsoner alright, I'd copy an existing example (style wise) then(?) and get back once I'm done(?) - Does it go into the "inverse" folder?

When I add dipy to the environment.yml (it's "pip-able"), I just add it to the pip list? and make a line in the README.rst in the "For full functionality, some functions require:" -section?

dengemann, 2018-05-16:
Hi Tommy,

Yes add something to inverse and just follow the style of other examples.
Don’t hesitate to push early and expose the main ideas. We can refine
things interactively.

dengemann, 2018-05-16:
And don’t bother about readme right now. Let’s first do code.

larsoner, 2018-12-12:
Closing as this is done!

closed 2018-12-12 by larsoner