mne-tools/mne-python #5210: Adding extra data to the info dictionary for Raw objects

monkeyman192, 2018-05-16:
I have noticed that there is a reasonable amount of information extracted from raw files (I am looking at KIT ones in particular) that isn't saved.
For example the system_name and model_name aren't used at all, but this information would be useful to store as it could be used by the mne-bids library to add extra data to the files produced.

agramfort, 2018-05-16:
the mne measurement info is quite strict in its content.

see:

https://martinos.org/mne/stable/generated/mne.Info.html
https://martinos.org/mne/stable/auto_tutorials/plot_info.html

can you see a way to store this info without changing the Info object?

agramfort, 2018-05-19:
what are the possible values of system_name? if there is a short list then
we could make a function in the base class.

I imaging something like “Vectorview" ...

larsoner, 2018-05-19:
With the proposed approach, doing `raw.save(...)` and `read_raw_fif(...)` will not work, the information will be lost.

I suggest that if possible instead we use these fields, if they aren't currently used by `read_raw_kit` -- here are some entries for the `sample` data:
```
>>> info['description']
'acquisition (megacq) VectorView system at NMR-MGH'
>>> info['experimenter']
'MEG'
>>> info['proj_name']
'test'
```
And here for some data I recorded at UW:
```
>>> info['description']
'Vectorview system'
>>> info['experimenter']
'neuromag'
>>> info['proj_name']
'eric_non_space'
```
And for our TRIUX test data:
```
>>> info['description']
'Lobier (muriel) TRIUX system'
>>> info['experimenter']
'Muriel'
>>> info['proj_name']
'visatt_tms_meg'
```
In particular it looks like `info['description']` tends to hold the site / system info.

agramfort, 2018-08-04:
@monkeyman192 can we close this?

monkeyman192, 2018-08-04:
Yep, sorry. This issue is essentially covered by [this](https://github.com/mne-tools/mne-bids/issues/47)

closed 2018-08-04 by agramfort