mne-tools/mne-python #5214: Import mayavi.mlab issue

d-van-de-velden, 2018-05-16:
Dear all,

I am using Spyder3 and Python2.7 and downloaded the newest mne-python package. 
Unfortunately  I am not able to import mayavi.mlab.
Code:
```  
import mne
import getpass
from pyface.qt import QtGui, QtCore
from mayavi import mlab
```

Error:

> Changing backend to Qt for Mayavi
> Traceback (most recent call last):
> 
>   File "<ipython-input-3-d0c45c8a837c>", line 11, in <module>
>     from mayavi import mlab
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/mayavi/mlab.py", line 27, in <module>
>     from mayavi.tools.camera import view, roll, yaw, pitch, move
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/mayavi/tools/camera.py", line 25, in <module>
>     from .engine_manager import get_engine
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/mayavi/tools/engine_manager.py", line 12, in <module>
>     from mayavi.preferences.api import preference_manager
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/mayavi/preferences/api.py", line 4, in <module>
>     from .preference_manager import preference_manager
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/mayavi/preferences/preference_manager.py", line 29, in <module>
>     from traitsui.api import View, Group, Item
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/traitsui/api.py", line 36, in <module>
>     from .editors.api import ArrayEditor
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/traitsui/editors/__init__.py", line 23, in <module>
>     from .api import ArrayEditor
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/traitsui/editors/api.py", line 43, in <module>
>     from .list_str_editor import ListStrEditor
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/traitsui/editors/list_str_editor.py", line 33, in <module>
>     from pyface.image_resource import ImageResource
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/pyface/image_resource.py", line 20, in <module>
>     from .toolkit import toolkit_object
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/pyface/toolkit.py", line 27, in <module>
>     toolkit = toolkit_object = find_toolkit('pyface.toolkits')
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/pyface/base_toolkit.py", line 263, in find_toolkit
>     return import_toolkit(ETSConfig.toolkit, entry_point)
> 
>   File "/home/dvelden/anaconda2/lib/python2.7/site-packages/pyface/base_toolkit.py", line 209, in import_toolkit
>     raise RuntimeError(msg)
> 
> RuntimeError: No pyface.toolkits plugin found for toolkit qt4

I need help with this. Installing/Uninstalling various packages wasn't succesful. Already uninstalled "anaconda-navigator". 

Thanks in advance for any help.

thesujitroy, 2018-05-16:
Hi!
You can create an environment in anaconda with py2.7.x and then try to install mayavi in that package it should work.

larsoner, 2018-05-16:
Can you try our now-standard `environment.yml` on the [install instructions](https://martinos.org/mne/stable/install_mne_python.html) page?

d-van-de-velden, 2018-05-16:
simply installing 'pyqt' and 'pyside' does not fix it. 
I am currently trying the now-standard environment....

d-van-de-velden, 2018-05-16:
Works fine in 'ipython' and 'python'. Sorry to bother you about it. Wanted it to work in Spyder3 aswell. It's not working their right now, but at least we got 'ipython' working

larsoner, 2018-05-16:
Spyder does not work for you in this environment? I have used it in macOS and Linux at least. What OS are you on?

d-van-de-velden, 2018-05-16:
Linux Ubuntu 16.04

Error:

> QOpenGLShaderProgram::uniformLocation(qt_Matrix): shader program is not linked

larsoner, 2018-05-16:
what does `$ glxinfo | grep core` give? Is it possible your graphics card is old?

larsoner, 2018-05-16:
What is `$ glxinfo | grep core`?

d-van-de-velden, 2018-05-16:
> $ glxinfo | grep core
> OpenGL core profile version string: 4.5.0 NVIDIA 384.111
> OpenGL core profile shading language version string: 4.50 NVIDIA
> OpenGL core profile context flags: (none)
> OpenGL core profile profile mask: core profile
> OpenGL core profile extensions:

larsoner, 2018-05-16:
Okay so you are indeed on the proprietary drivers. Can you try doing `$ pip install --upgrade pyqt5>=5.10` to see if it helps? It's possible that this is also needed on some Linux systems in addition to macOS.

d-van-de-velden, 2018-05-17:
This just changed the Error to:

> (mne) dvelden@megpowerpc:~$ spyder
> [0517/093918.649611:WARNING:stack_trace_posix.cc(648)] Failed to open file: /tmp/.glgKR2k4 (deleted)
>   Error: No such file or directory
> Could not find QtWebEngineProcess
> [23954:23954:0517/093918.789278:FATAL:zygote_host_impl_linux.cc(182)] Check failed: ReceiveFixedMessage(fds[0], kZygoteBootMessage, sizeof(kZygoteBootMessage), &boot_pid). 
> #0 0x7fdb53b8e0a5 <unknown>
> #1 0x7fdb53ba3eca <unknown>
> #2 0x7fdb531e6fc3 <unknown>
> #3 0x7fdb531e5c3b <unknown>
> #4 0x7fdb531e6219 <unknown>
> #5 0x7fdb52e2993a <unknown>
> #6 0x7fdb52e2ee76 <unknown>
> #7 0x7fdb5294c600 <unknown>
> #8 0x7fdb5294db25 <unknown>
> #9 0x7fdb5293cee9 <unknown>
> #10 0x7fdb5293d0be QtWebEngineCore::WebContentsAdapter::WebContentsAdapter()
> #11 0x7fdb5888e441 <unknown>
> #12 0x7fdb5888ef24 QWebEnginePage::QWebEnginePage()
> #13 0x7fdb58ad0229 sipQWebEnginePage::sipQWebEnginePage()
> #14 0x7fdb58ad03e2 init_type_QWebEnginePage
> #15 0x7fdb5c1c0a74 sipSimpleWrapper_init
> #16 0x563ffedee9b7 type_call
> #17 0x563ffed5ed7b _PyObject_FastCallDict
> #18 0x563ffedee7ce call_function
> #19 0x563ffee10cba _PyEval_EvalFrameDefault
> #20 0x563ffede8d7b _PyFunction_FastCallDict
> #21 0x563ffed5ef5f _PyObject_FastCallDict
> #22 0x563ffed63a03 _PyObject_Call_Prepend
> #23 0x563ffed5e99e PyObject_Call
> #24 0x563ffedbb02b slot_tp_init
> #25 0x563ffedee9b7 type_call
> #26 0x563ffed5ed7b _PyObject_FastCallDict
> #27 0x563ffedee7ce call_function
> #28 0x563ffee10cba _PyEval_EvalFrameDefault
> #29 0x563ffede8d7b _PyFunction_FastCallDict
> #30 0x563ffed5ef5f _PyObject_FastCallDict
> #31 0x563ffed63a03 _PyObject_Call_Prepend
> #32 0x563ffed5e99e PyObject_Call
> #33 0x563ffedbb02b slot_tp_init
> #34 0x563ffedee9b7 type_call
> #35 0x563ffed5ed7b _PyObject_FastCallDict
> #36 0x563ffedee7ce call_function
> #37 0x563ffee10cba _PyEval_EvalFrameDefault
> #38 0x563ffede8d7b _PyFunction_FastCallDict
> #39 0x563ffed5ef5f _PyObject_FastCallDict
> #40 0x563ffed63a03 _PyObject_Call_Prepend
> #41 0x563ffed5e99e PyObject_Call
> #42 0x563ffedbb02b slot_tp_init
> #43 0x563ffedee9b7 type_call
> #44 0x563ffed5ed7b _PyObject_FastCallDict
> #45 0x563ffedee7ce call_function
> #46 0x563ffee10cba _PyEval_EvalFrameDefault
> #47 0x563ffede7dae _PyEval_EvalCodeWithName
> #48 0x563ffede8e1b _PyFunction_FastCallDict
> #49 0x563ffed5ef5f _PyObject_FastCallDict
> #50 0x563ffed63a03 _PyObject_Call_Prepend
> #51 0x563ffed5e99e PyObject_Call
> #52 0x563ffedbb02b slot_tp_init
> #53 0x563ffedee9b7 type_call
> #54 0x563ffed5ed7b _PyObject_FastCallDict
> #55 0x563ffedee7ce call_function
> #56 0x563ffee10cba _PyEval_EvalFrameDefault
> #57 0x563ffede7dae _PyEval_EvalCodeWithName
> #58 0x563ffede8941 fast_function
> #59 0x563ffedee755 call_function
> #60 0x563ffee10cba _PyEval_EvalFrameDefault
> #61 0x563ffede870b fast_function
> 
> Received signal 6
> #0 0x7fdb53b8e0a5 <unknown>
> #1 0x7fdb528cf141 <unknown>
> #2 0x7fdb53b8e43d <unknown>
> #3 0x7fdb7be9f390 <unknown>
> #4 0x7fdb7baf9428 gsignal
> #5 0x7fdb7bafb02a abort
> #6 0x7fdb53b8ddc5 <unknown>
> #7 0x7fdb53ba41ed <unknown>
> #8 0x7fdb531e6fc3 <unknown>
> #9 0x7fdb531e5c3b <unknown>
> #10 0x7fdb531e6219 <unknown>
> #11 0x7fdb52e2993a <unknown>
> #12 0x7fdb52e2ee76 <unknown>
> #13 0x7fdb5294c600 <unknown>
> #14 0x7fdb5294db25 <unknown>
> #15 0x7fdb5293cee9 <unknown>
> #16 0x7fdb5293d0be QtWebEngineCore::WebContentsAdapter::WebContentsAdapter()
> #17 0x7fdb5888e441 <unknown>
> #18 0x7fdb5888ef24 QWebEnginePage::QWebEnginePage()
> #19 0x7fdb58ad0229 sipQWebEnginePage::sipQWebEnginePage()
> #20 0x7fdb58ad03e2 init_type_QWebEnginePage
> #21 0x7fdb5c1c0a74 sipSimpleWrapper_init
> #22 0x563ffedee9b7 type_call
> #23 0x563ffed5ed7b _PyObject_FastCallDict
> #24 0x563ffedee7ce call_function
> #25 0x563ffee10cba _PyEval_EvalFrameDefault
> #26 0x563ffede8d7b _PyFunction_FastCallDict
> #27 0x563ffed5ef5f _PyObject_FastCallDict
> #28 0x563ffed63a03 _PyObject_Call_Prepend
> #29 0x563ffed5e99e PyObject_Call
> #30 0x563ffedbb02b slot_tp_init
> #31 0x563ffedee9b7 type_call
> #32 0x563ffed5ed7b _PyObject_FastCallDict
> #33 0x563ffedee7ce call_function
> #34 0x563ffee10cba _PyEval_EvalFrameDefault
> #35 0x563ffede8d7b _PyFunction_FastCallDict
> #36 0x563ffed5ef5f _PyObject_FastCallDict
> #37 0x563ffed63a03 _PyObject_Call_Prepend
> #38 0x563ffed5e99e PyObject_Call
> #39 0x563ffedbb02b slot_tp_init
> #40 0x563ffedee9b7 type_call
> #41 0x563ffed5ed7b _PyObject_FastCallDict
> #42 0x563ffedee7ce call_function
> #43 0x563ffee10cba _PyEval_EvalFrameDefault
> #44 0x563ffede8d7b _PyFunction_FastCallDict
> #45 0x563ffed5ef5f _PyObject_FastCallDict
> #46 0x563ffed63a03 _PyObject_Call_Prepend
> #47 0x563ffed5e99e PyObject_Call
> #48 0x563ffedbb02b slot_tp_init
> #49 0x563ffedee9b7 type_call
> #50 0x563ffed5ed7b _PyObject_FastCallDict
> #51 0x563ffedee7ce call_function
> #52 0x563ffee10cba _PyEval_EvalFrameDefault
> #53 0x563ffede7dae _PyEval_EvalCodeWithName
> #54 0x563ffede8e1b _PyFunction_FastCallDict
> #55 0x563ffed5ef5f _PyObject_FastCallDict
> #56 0x563ffed63a03 _PyObject_Call_Prepend
> #57 0x563ffed5e99e PyObject_Call
> #58 0x563ffedbb02b slot_tp_init
> #59 0x563ffedee9b7 type_call
> #60 0x563ffed5ed7b _PyObject_FastCallDict
> #61 0x563ffedee7ce call_function
>   r8: 00007fdb7be8a770  r9: 00007fdb7c2a8700 r10: 0000000000000008 r11: 0000000000000202
>  r12: 00007ffef4d80d00 r13: 00000000000000ac r14: 00007ffef4d80f30 r15: 00007ffef4d816a0
>   di: 0000000000005d92  si: 0000000000005d92  bp: 00007fdb784d7340  bx: 00007ffef4d81480
>   dx: 0000000000000006  ax: 0000000000000000  cx: 00007fdb7baf9428  sp: 00007ffef4d80b98
>   ip: 00007fdb7baf9428 efl: 0000000000000202 cgf: 002b000000000033 erf: 0000000000000000
>  trp: 0000000000000000 msk: 0000000000000000 cr2: 0000000000000000
> [end of stack trace]
> Calling _exit(1). Core file will not be generated.

danieltomasz, 2018-06-19:
@d-van-de-velden did you tried latest  4.6 mayavi release (from end of may)? I also have  problem with mlab but with the new release everything is working

drammock, 2020-11-24:
closing after >2 years of no response.

closed 2020-11-24 by drammock