mne-tools/mne-python #5217: Bad ticks in colorbar for evoked.topomap

sappelhoff, 2018-05-18:
When using `evoked.plot_topomap()`, I consistently run into two problems:

1. The colorbar ticks are "incomplete" in that the ticks cover only half of the bar or stop prematurely (i.e., without providing a label for the maximum and minimum)

2. When using the `vmin` and `vmax` parameters and `np.abs(vmin) != np.abs(vmax)`, the zero point is not "white" in terms of the colormap.

# Minimal code example
```python
import os
import mne

# Get data paths
sample_path = mne.datasets.sample.data_path()
sample_data_path = os.path.join(sample_path, 'MEG', 'sample', 'sample_audvis_raw.fif')
sample_events_path = os.path.join(sample_path, 'MEG', 'sample', 'sample_audvis_raw-eve.fif')

# Load raw and events
raw = mne.io.read_raw_fif(sample_data_path, preload=True)
events = mne.read_events(sample_events_path)

# Epoch and evoke
raw.pick_types(meg=False, eeg=True)
epochs = mne.Epochs(raw, events, event_id=dict(foo=4))
evoked = epochs.average()

# Plot it
evoked.plot_topomap(times=np.arange(0,0.5,0.1),
                    time_unit='ms', ch_type='eeg',
                    vmin=-15, vmax=15, colorbar=True)    
```

# Illustrates problem 1
```
evoked.plot_topomap(times=np.arange(0,0.5,0.1),
                    time_unit='ms', ch_type='eeg',
                    vmin=-15, vmax=15, colorbar=True)    
```
[image]

# Illustrates problem 2 (note vmin and vmax)
```
evoked.plot_topomap(times=np.arange(0,0.5,0.1),
                    time_unit='ms', ch_type='eeg',
                    vmin=-7, vmax=3, colorbar=True)    
```
[image]

agramfort, 2018-05-19:
it's a matplotlib bug. I fixed it by :

pip install matplotlib==2.0.2

sappelhoff, 2018-05-22:
great, thanks @agramfort

closed 2018-05-22 by sappelhoff