mne-tools/mne-python #5223: Spliting the inverse solution of mixed source space into surface and volume space

poojaprabhu9, 2018-05-22:
How can the inverse solution( or "stcs") of mixed source space can be split into inverse solution of surface surce space and inverse solution of volume source space seperately. So that these two can be visualised seperately over MRI?
please find the code.

[plot_mixed_source_space_inverse.docx](https://github.com/mne-tools/mne-python/files/2028248/plot_mixed_source_space_inverse.docx)

agramfort, 2018-05-24:
@poojaprabhu9 please don't put code on .docx files but like I did above or using https://gist.github.com/

basically the need is to go from the stcs above into 2 stcs one for the surface par and one for the volume part.

I would go with:

```
stc_surf, stc_vol = stcs.split(src)
```
for the background the objective is to include deep structures in MNE inverse and be able to get a surface stc that can be visualized with stc.plot

any volunteer to help? maybe @SherazKhan ?

poojaprabhu9, 2018-05-24:
Sure @agramfort . I will try with that.
Thanks

larsoner, 2021-01-07:
We have `.volume` and `.surface` methods nowadays, so this should be fixed

closed 2021-01-07 by larsoner