mne-tools/mne-python #5224: problem in visualising the inverse source solution when more than one volume labels are used

poojaprabhu9, 2018-05-22:
How can the code can be modified to visualize the inverse solution of volume source space on MRI
 when two volume labels are given. 
For example in the code give: 
labels_vol = ['Left-Hippocampus', 'Right-Hippocampus']

[visualization of volume source space on MRI.docx](https://github.com/mne-tools/mne-python/files/2028419/visualization.of.volume.source.space.on.MRI.docx)

agramfort, 2018-05-24:
@poojaprabhu9 please edit the issue as I did here https://github.com/mne-tools/mne-python/issues/5223 to avoid the .docx file. Tell us when it's done.

poojaprabhu9, 2018-05-24:
@agramfort . Sorry for Docx file. Here is the code. 
'''
import numpy as np  
import os.path as op
import matplotlib.pyplot as plt
import mne
from mne.datasets import sample
from mne import setup_volume_source_space, setup_source_space
from mne import make_forward_solution
from mne.minimum_norm import make_inverse_operator, apply_inverse
from nilearn.plotting import plot_stat_map
from nilearn.image import index_img
# Set dir
data_path = sample.data_path()
subject = 'sample'
data_dir = op.join(data_path, 'MEG', subject)
subjects_dir = op.join(data_path, 'subjects')
bem_dir = op.join(subjects_dir, subject, 'bem')
# Set file names
fname_aseg = op.join(subjects_dir, subject, 'mri', 'aseg.mgz')
fname_model = op.join(bem_dir, '%s-5120-bem.fif' % subject)
fname_bem = op.join(bem_dir, '%s-5120-bem-sol.fif' % subject)
fname_evoked = data_dir + '/sample_audvis-ave.fif'
fname_trans = data_dir + '/sample_audvis_raw-trans.fif'
fname_fwd = data_dir + '/sample_audvis-meg-oct-6-mixed-fwd.fif'
fname_cov = data_dir + '/sample_audvis-shrunk-cov.fif'
#########################################################################
labels_vol = ['Left-Hippocampus']  # labels_vol = ['Left-Hippocampus', ‘Right-Hippocampus’]
sphere=(0,0,0,120)
src = setup_volume_source_space(subject, mri=fname_aseg, pos=2.0, sphere=sphere,
                                        volume_label=labels_vol, subjects_dir=subjects_dir, 
                                        verbose=True)
fwd = make_forward_solution(fname_evoked, fname_trans, src, fname_bem, mindist=5.0,  
                            meg=True, eeg=False, n_jobs=1)
src_fwd = fwd['src']
n = sum(src_fwd[i]['nuse'] for i in range(len(src_fwd)))
print('the fwd src space contains %d spaces and %d points' % (len(src_fwd), n))
# Compute inverse solution 
snr = 3.0           
inv_method = 'MNE' # or MNE
lambda2 = 1.0 / snr ** 2
# Load data
condition = 'Left Auditory'
evoked = mne.read_evokeds(fname_evoked, condition=condition,
                          baseline=(None, 0))
noise_cov = mne.read_cov(fname_cov)
# Compute inverse operator
inverse_operator = make_inverse_operator(evoked.info, fwd, noise_cov,
                                         depth=None, fixed=False)
stcs = apply_inverse(evoked, inverse_operator, lambda2, inv_method,
                     pick_ori=None)
######################################################################
src = fwd['src']
img = stcs.as_volume(src, mri_resolution=False)  # set True for full MRI resolution
t1_fname = data_path + '/subjects/sample/mri/T1.mgz'
# Plotting with nilearn ######################################################
plot_stat_map(index_img(img, 200), t1_fname, threshold=2*1e-9,
              title='%s (t=%.1f s.)' % (inv_method, stcs.times[61]))
plt.show()
,,,

agramfort, 2018-10-06:
we have a regression on this with the new SourceMorph PR :(

see this gist https://gist.github.com/agramfort/394e0155945261ccf3faeea0f66d8dbe
to replicate the problem.

Basically we now again only take the first label from the VolSourceEstimate ...

cc @poojaprabhu9

PR #5231 (2018-06-28): MRG : better detect type of src for making an stc
closes #5224 

to debug

```
import os.path as op
import matplotlib.pyplot as plt
import mne
from mne.datasets import sample
from mne import setup_volume_source_space
from mne import make_forward_solution
from mne.minimum_norm import make_inverse_operator, apply_inverse
from nilearn.plotting import plot_stat_map
from nilearn.image import index_img

# Set dir
data_path = sample.data_path()
subject = 'sample'
data_dir = op.join(data_path, 'MEG', subject)
subjects_dir = op.join(data_path, 'subjects')
bem_dir = op.join(subjects_dir, subject, 'bem') [... cut]
PR #5590 (2018-10-09): [MRG+1] fix morph of volume source space made of more than 1 label
fixes #5224

_interpolate_data in morph.py was only taking into account the first label.

now it's looping properly over all src label.

let me know if you see a way to test this.

@poojaprabhu9 please confirm if this fixes your problem.
closed 2018-10-09 by agramfort