mne-tools/mne-python #5229: allow to pass a cov to beamformer filters to get a static image as in FieldTrip

agramfort, 2018-05-24:
cc @britta-wstnr @wmvanvliet

britta-wstnr, 2018-05-24:
giving back a power estimate? We have ``_lcmv_source_power`` which could be changed accordingly. I think I even started that somewhere and can have a look soon.
It is in the part that is not touched by https://github.com/mne-tools/mne-python/pull/5135 so it shouldn't clash with that one either.

vlitvak, 2018-05-24:
I must say I'm only getting to grips with MNE code structure but to me it seems that a sensible way to implement this would be as apply_lcmv_covariance that would take covariance as an input and output a source time course with a single frame.

robertoostenveld, 2018-05-24:
note that the units would be power rather than amplitude, and the values cannot be negative, whereas the source time (normally) could

wmvanvliet, 2018-05-24:
The implementation can be mostly taken from our `apply_dics_csd` method.

PR #6227 (2019-05-02): MRG: Power estimate for LCMV: apply_lcmv_cov
This allows to apply the LCMV spatial filter to the covariance matrix to obtain a power estimate (not time-resolved).
In the process, we refactored ``apply_dics_csd`` with @wmvanvliet to have one function ``_compute_power``, since the operations for LCMV and DICS are similar.

A test against FieldTrip is to follow, output converges.

Closes #5229
closed 2019-05-02 by larsoner