mne-tools/mne-python #5234: Wrong title layout when calling historgram

massich, 2018-05-28:
When calling histogram by `b` in a raw plot the axis labels do not render properly. 

```
import mne
data_path = mne.datasets.sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname)
raw.plot()
```

jona-sassenhagen, 2018-05-28:
Sorry, what is the problem? That it needs a somewhat diff. layout?

massich, 2018-05-28:
The space between the plots does not allow to read **uV** and **fT** which ends underneath **Magnetometers** and **Gradiometers**

agramfort, 2018-05-28:
layout need to be adjusted in
https://github.com/mne-tools/mne-python/blob/master/mne/viz/epochs.py#L1926

@massich can you take care of it?

PR #5247 (2018-05-31): [MRG] fix overlaping subplots in evoked histograms
fixes #5234 
[image]
closed 2018-05-31 by agramfort