mne-tools/mne-python #5237: Add spectra to report function?

jona-sassenhagen, 2018-05-28:
Maybe it would make sense to add spectra to the reports. They can be very informative.

Another idea would be epochs images, which can also be highly diagnostic.

Not volunteering :)

@jasmainak

jasmainak, 2018-05-28:
I agreee ;)

dengemann, 2018-05-28:
Yes @jona-sassenhagen @jasmainak this is a good idea. At least for the power spectra :)

agramfort, 2018-08-04:
@jona-sassenhagen still not volunteering?

jona-sassenhagen, 2018-08-04:
As you see, I just opened 3 other PRs instead 8)

larsoner, 2018-08-04:
Given how easy it is to manually add `matplotlib` and `mayavi` figures using the `add_*` functions, I'm not sure we should add too many of these. PSD calculation on raw can take a while, too. But I agree this is almost always useful, so on the balance I'm +0.5 on adding it. I might have time to try

agramfort, 2018-08-05:
yes it takes time but I typically compute it on the first minute of data
and not the full recording. we should do this by default to limit computing
cost.

larsoner, 2018-08-05:
> yes it takes time but I typically compute it on the first minute of data
> and not the full recording. we should do this by default to limit computing
> cost.

I disagree. The `Report` function should follow the defaults of `raw.plot_psd`, otherwise it's unexpected behavior. The way I coded it, it's easy for you to change this behavior during rendering if you want. FWIW the default of `plot_psd` actually originally had a limited `tmax` but we decided to change this at some point to use the entire recording.

agramfort, 2018-08-05:
ok makes sense to follow default of plot_psd

PR #5403 (2018-08-13): MRG: Add raw PSD to parse_folder
Closes #5237.

Also shifts our docs from `mne.report` namespace to the root namespace as `mne.Report` already lives there.

Copying the truncated testing raw file to desktop and doing:
```
import mne
report = mne.Report(raw_psd=True)
report.parse_folder('/Users/larsoner/Desktop')
report.save('/Users/larsoner/Desktop/report.html')
```
I get:

[image]

@jona-sassenhagen happy?
closed 2018-08-13 by jasmainak