mne-tools/mne-python #5244: plot_forward_sensitivity_maps.py demo crashes with unexpected keyword

dnacombo, 2018-05-30:
In a fresh install from this morning, plot_forward_sensitivity_maps.py crashes with the following log:

```
================================================
Display sensitivity maps for EEG and MEG sensors
================================================

Sensitivity maps can be produced from forward operators that
indicate how well different sensor types will be able to detect
neural currents from different regions of the brain.

To get started with forward modeling see :ref:`tut_forward`.

Reading forward solution from /lena13/home_users/users/datalinks/CAUSALITY_JUELICH/Collab_Juelich_CTPS+Causality_Data/Analysis_Max/MNE_analysis/MNE-sample-data/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif...
    Reading a source space...
    Computing patch statistics...
    Patch information added...
    Distance information added...
    [done]
    Reading a source space...
    Computing patch statistics...
    Patch information added...
    Distance information added...
    [done]
    2 source spaces read
    Desired named matrix (kind = 3523) not available
    Read MEG forward solution (7498 sources, 306 channels, free orientations)
    Desired named matrix (kind = 3523) not available
    Read EEG forward solution (7498 sources, 60 channels, free orientations)
    MEG and EEG forward solutions combined
    Source spaces transformed to the forward solution coordinate frame
    Average patch normals will be employed in the rotation to the local surface coordinates....
    Converting to surface-based source orientations...
    [done]
Leadfield size : 366 x 22494
    204 out of 366 channels remain after picking
    102 out of 366 channels remain after picking
    60 out of 366 channels remain after picking
Adding average EEG reference projection.
/home/anaconda3/envs/mne/lib/python3.6/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
Traceback (most recent call last):

  File "<ipython-input-1-1b404498cb31>", line 1, in <module>
    runfile('/lena13/home_users/users/chaumon/owncloud/Lab/Projects/SOLID_MEEG/code/py/plot_forward_sensitivity_maps.py', wdir='/lena13/home_users/users/chaumon/owncloud/Lab/Projects/SOLID_MEEG/code/py')

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/lena13/home_users/users/chaumon/owncloud/Lab/Projects/SOLID_MEEG/code/py/plot_forward_sensitivity_maps.py", line 71, in <module>
    clim=dict(lims=[0, 50, 100]), smoothing_steps = 1)

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/mne/source_estimate.py", line 1463, in plot
    spacing=spacing)

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/mne/viz/_3d.py", line 1669, in plot_source_estimates
    views=views)

TypeError: __init__() got an unexpected keyword argument 'curv'

```

Removing that `curv=True` argument altogether in `_3d.py` lets it run for me.

agramfort, 2018-05-30:
are you using version 0.16.1 ? how did you install? is your PySurfer
install up to date?

dnacombo, 2018-05-30:
Installed following instructions here [https://martinos.org/mne/stable/install_mne_python.html#install-dependencies-and-mne](url).
version is 0.16.1

There was a warning about 
```
mkl-random 1.0.1 requires cython, which is not installed.
mkl-fft 1.0.0 requires cython, which is not installed.
```
Installing the missing packages and starting in new terminal does not fix it.
Right now:
```
pip install -U pysurfer
Requirement already up-to-date: pysurfer in /home/anaconda3/envs/mne/lib/python3.6/site-packages (0.9.dev0)
Requirement not upgraded as not directly required: nibabel>=1.2 in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from pysurfer) (2.2.1)
Requirement not upgraded as not directly required: mayavi in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from pysurfer) (4.6.1.dev0)
Requirement not upgraded as not directly required: pygments in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (2.2.0)
Requirement not upgraded as not directly required: envisage in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (4.6.0)
Requirement not upgraded as not directly required: vtk in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (8.1.0)
Requirement not upgraded as not directly required: traits>=4.6.0 in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (4.6.0)
Requirement not upgraded as not directly required: pyface>=6.0.0 in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (6.0.0)
Requirement not upgraded as not directly required: apptools in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (4.4.0)
Requirement not upgraded as not directly required: traitsui>=6.0.0 in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (6.0.0)
Requirement not upgraded as not directly required: numpy in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from mayavi->pysurfer) (1.14.3)
Requirement not upgraded as not directly required: configobj in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from apptools->mayavi->pysurfer) (5.0.6)
Requirement not upgraded as not directly required: six in /home/anaconda3/envs/mne/lib/python3.6/site-packages (from configobj->apptools->mayavi->pysurfer) (1.11.0)
```

agramfort, 2018-05-30:
indeed I see a curv=True here in 0.16 branch:

https://github.com/mne-tools/mne-python/blob/maint/0.16/mne/viz/_3d.py#L1665

but I don't see any curv parameter passed on current master:

https://github.com/mne-tools/mne-python/blob/master/mne/viz/_3d.py#L1665

@larsoner did we break backward compat with pysurfer?

larsoner, 2018-05-30:
Need to backport #5216.

In the meantime we can also fix by pinning the PySurfer version, which is probably a good idea anyway.

larsoner, 2018-05-30:
Backported, and un-pinning PR opened

larsoner, 2018-05-30:
@dnacombo to fix your local issue you can do:
```
$ pip uninstall PySurfer
$ pip install PySurfer
```

larsoner, 2018-05-30:
Argh @agramfort looking at it more closely, some of the stuff for vector source estimates still won't quite work properly. So @dnacombo instead of what I suggested, you probably need to locally update MNE to 0.16.2 instead:
```
pip install --upgrade --no-deps "https://api.github.com/repos/mne-tools/mne/zipball/maint/0.16"
```

closed 2018-05-30 by larsoner

dnacombo, 2018-05-30:
`error 404 Client Error: Not Found for url: https://api.github.com/repos/mne-tools/mne/zipball/maint/0.16`

larsoner, 2018-05-30:
Can you try `https://github.com/mne-tools/mne-python/archive/maint/0.16.zip` as the URL?

dnacombo, 2018-05-30:
This is installing 0.16.1. And the script now works.

agramfort, 2018-05-30:
we'll do a bug fix release asap