mne-tools/mne-python #5249: Brainvision stim channel creation fails if event durations are 0

jona-sassenhagen, 2018-05-30:
Stim channels for brainvision and EEGLAB files are created like this:

```
    for onset, duration, trigger in events:
        stim_channel[onset:onset + duration] = trigger
```

If the duration is 0, which it often is, this will result in no events being recorded.

I am not sure how to deal with this.

The events are still available to the user at `raw._events`, but `mne.find_events` etc won't work.

larsoner, 2018-05-30:
we should probably make the min duration 1 sample

jona-sassenhagen, 2018-05-30:
BV vmrk files have a duration column that is set to 0 for the file I'm looking at ... but yeah, we can do `min(duration, 1)` in _create_stim_channel`

mmagnuski, 2018-06-01:
I've just had somebody contact me with this problem (as it turned out) - I'm definitely +1 on Jona's proposal.

agramfort, 2018-06-02:
agreed

jona-sassenhagen, 2018-06-10:
@agramfort ... and you actually said "agreed" here :)

agramfort, 2018-06-10:
yes there is no cleaner solution now.

PR #5274 (2018-06-12): MRG Min duration for stim channel synthesis = 1
closes #5249
closed 2018-06-12 by agramfort