mne-tools/mne-python #5258: CTF meas_id

bloyl, 2018-06-04:
Not sure if this is related to #4990, but I'm running into a problem with meas_ids not serializing correctly.

To replicate.
```
import os.path as op
from mne.io import read_raw_ctf, read_raw_fif
from mne.datasets import testing
import tempfile

ctf_dir = op.join(testing.data_path(download=False), 'CTF')
ctf_fname_somato = op.join(ctf_dir, 'testdata_ctf_short.ds')

raw_orig = read_raw_ctf(ctf_fname_somato)

temp_file = tempfile.NamedTemporaryFile(suffix='-raw.fif')
raw_orig.save(temp_file.name, overwrite=True)
raw_fif = read_raw_fif(temp_file.name)

for k in ['secs', 'usecs']:
    print('orig %s - %s' % (k, raw_orig.info['meas_id'][k]))
    print('fif  %s - %s' % (k, raw_fif.info['meas_id'][k]))
    print('')
```

larsoner, 2018-06-04:
If you round-trip again do you at least get the same ID that time?

Can you try `git bisect`? First might be to try one commit before that PR was merged (f5e3399ad6d0c163f3b86f002fd3c6fb687e6315) and then the commit where it was merged (d77cc95b0cf56b1df5886c9ebc52261a55695331).

bloyl, 2018-06-04:
round-tripping a second time works. Meaning that saving the fif file and reading it back in results in the same meas_id structure.

My guess is that the ctf reader, specifically:
https://github.com/mne-tools/mne-python/blob/ab4571a83c4e5d5de0dcb2e9346f809f113dd619/mne/io/ctf/info.py#L420-L423

is a different convention (ie packing everything in secs instead of usecs?) then what is used for fif files?

Let me know if you still want me to mess with `git bisect`. I might have some time tonight to try it out.

larsoner, 2018-06-04:
It's still worth doing at least `git checkout f5e3399` (before that PR was merged) and then running these tests again. Based on what you say I think the result will stay the same, namely that the problem existed before that PR and is probably a CTF-related bug.

bloyl, 2018-06-05:
This is a bug that seems to effect all IO not just CTF

```
import os.path as op
from mne.io import read_raw_fif
from mne.datasets import testing
import tempfile

data_dir = op.join(testing.data_path(download=False), 'MEG','sample')
fif_fname = op.join(data_dir, 'sample_audvis_trunc_raw.fif')
raw_orig = read_raw_fif(fif_fname)

temp_file = tempfile.NamedTemporaryFile(suffix='-raw.fif')
raw_orig.save(temp_file.name, overwrite=True)

raw_fif = read_raw_fif(temp_file.name)

for k in ['secs', 'usecs']:
    print('orig %s - %s' % (k, raw_orig.info['meas_id'][k]))
    print('fif  %s - %s' % (k, raw_fif.info['meas_id'][k]))
    print('')
```
 yields
```
orig secs - 1411146352
fif  secs - 0

orig usecs - 0
fif  usecs - 2147483647
```
--------------------------------------------------------------------------------------------------------------
my guess is that this code
https://github.com/mne-tools/mne-python/blob/ab4571a83c4e5d5de0dcb2e9346f809f113dd619/mne/io/meas_info.py#L1219-L1231
needs to be updated post #4990 

--------------------------------------------------------------------------------------------------------------
actually if i do git checkout f5e3399
the above code still give something strange...
```
orig secs - 1411146352
fif  secs - 1528209476

orig usecs - 0
fif  usecs - 88141
```

Am I wrong to expect these to be the same?

larsoner, 2018-06-05:
Actually I think this was the intended behavior of #4990. As an anonymization step it's better not to save the date when the file is written. So `secs==0, usecs==2147483647` is a shorthand for `None` (since you would never in practice write `usecs >= 1e6` in this format)

bloyl, 2018-06-05:
> usecs==2147483647 <(214)%20748-3647> is a shorthand for None (since you

ok so its safe to assume that secs and usecs in meas_id are not preserved
through file IO. So I can remove that part of the test from the
io/ctf/test/test_ctf.py?

larsoner, 2018-06-05:
Better yet, ensure it's equal to `DATE_NONE`

agramfort, 2018-06-05:
if you don't call anonymize I would not expect any change of the meas_id

do I miss something?

larsoner, 2018-06-05:
You're right, `DATE_NONE` should only be used when the date is unknown, or when anonymizing.

So it seems like there is some bug. @bloyl can you see if just doing `read_info(...)` followed by `write_info(...)` has this problem of setting `info['meas_id'] == DATE_NONE`? It will help isolate the problem.

bloyl, 2018-06-05:
I added a fix and tests to #5255 so lets move the conversation there.

closed 2018-07-23 by bloyl