mne-tools/mne-python #5265: Error in _check_compensation_grade for ICA on EEG data after switching from v15 to v16

SophieHerbst, 2018-06-06:
Dear all,
a script that ran under Version 15 now throws an error for every subject I tried. I am computing ICAs on epoched MEG and EEG data separately. The error occurs only for the EEG data. 

The error: 
```
File "/home/sh252228/anaconda2/lib/python2.7/site-packages/mne/utils.py", line 2213, in _check_compensation_grade
    name2, inst2.compensation_grade))
*** TypeError: %d format: a number is required, not NoneType
```

I realized that inst.compensation_grade is empty, while inst2.compensation_grade = 0

A snippet of my code:

```
ica_parameters = dict(n_components = .9999, method = 'fastica',  decim = 3, random_state = 23)
nip='az180202'

 # load data, hp filtered for ICA
 filename = root_directory + '/filtered/' + nip + '/' + nip + '_' + 'filtered4ICA_tsss_epo.fif'
 ICAData =  mne.read_epochs(filename, preload=True)
 
 ###################### run ICA ########################         
 picks_meg = mne.pick_types(ICAData.info, meg=True, eeg=False)
 picks_eeg = mne.pick_types(ICAData.info, meg=False, eeg=True)
     
  all_picks = dict({'meg':picks_meg,'eeg':picks_eeg})
     
  # Compute ICA, separately for EEG and MEG data 
  for ch_type in ['eeg']: # 'meg',
        print(ch_type)
        picks = all_picks[ch_type]
        
        # Apply ICA
        ica = ICA(n_components=ica_parameters['n_components'], 
                 method=ica_parameters['method'], random_state=ica_parameters['random_state'])
        print(ica)
            
        # Fit it to the data 
        ica.fit(ICAData, picks=picks, decim=ica_parameters['decim'], 
                reject=reject[ch_type])
```

I uploaded the data here:

agramfort, 2018-06-07:
@SophieHerbst is it with current master or version 0.16.1 ?

SophieHerbst, 2018-06-07:
0.16.1, yes
python version is 2.7.14

cbrnr, 2018-06-07:
@SophieHerbst what is the `reject` object you are passing as `reject=reject[ch_type]`?

cbrnr, 2018-06-07:
But yes, I can reproduce this error. No idea what the compensation grade is, but it seems to be something MEG-specific, so it should not be checked for EEG data...

SophieHerbst, 2018-06-07:
reject=dict(meg = dict(mag=10e-12, grad=500e-12), 
                eeg = dict(eeg=3000e-6))

Sorry, forgot that.

PR #5270 (2018-06-15): Fix check compensation on ica with EEG
closes #5265.

The problem arises because the ica object inherits the `info` of the object used to fit it with supplied 'picks' applied. however to avoid making a copy of time domain data the picks are never applied to the supplied object. This causes a discrepancy between the channels in `ica.info` and `raw.info` (or `epoch.info`) and the check fails.

I modified the `_check_compensation_grade` to accept channel names so the infos of both objects are mapped to the same channels before computing `compensation_grade`.

one case not tested is a ctf/eeg test [... cut]
closed 2018-06-15 by larsoner