mne-tools/mne-python #5266: Strange results after computing lcmv in volume source space

lbailey25, 2018-06-06:
Dear MNE users,

I'm attempting to compute lcmv beamformer on evoked data from a simple motor (finger tapping) task. I can compute and plot surface-based estimates with no problem, which gives me nice, sensible-looking activation maps. However, when I try to compute the stc in volume source space I get very bizarre results with a clear, circumscribed border of activation (see attached picture). The attached picture was generated using nilearn plot_stat_map, however I don't think it's a plotting issue, because when I export the stc as a .nii file I see the same thing in FSL.

[image]

Has anyone encountered this issue before? any advice would be appreciated!

larsoner, 2018-06-06:
@britta-wstnr any ideas?

britta-wstnr, 2018-06-07:
Hi @lbailey25 
Did you contrast conditions in your pipeline and/or use some kind of weight normalization (``weight_norm='unit-noise-gain'`` or ``'nai'``)?
Without knowing more about it your results look like a center-of-head bias.

lbailey25, 2018-06-07:
Hi @britta-wstnr 
Thanks for the reply. No, I didn't contrast conditions or use weight normalization. Below is the code that I used - perhaps it will shed some light on the problem
```
# Compute volume source space:
vol_src = setup_volume_source_space(subject=subject, bem=fname_bem, surface=None,
                                    mri=fname_aseg, mindist=5.0, exclude=0.0)

# Compute forward solution:
vol_fwd = fwd = make_forward_solution(fname_evoked, fname_trans, vol_src, fname_bem, mindist=0, meg=True, eeg=False)

# stc in volume source space:
filters = make_lcmv(evoked.info, vol_fwd, data_cov, noise_cov=noise_cov, pick_ori=None)
vol_stc = apply_lcmv(evoked, filters)
vol_stc = vol_stc.as_volume(src = fwd['src'], dest='mri')  
```
Hope that provides some useful information!

larsoner, 2019-03-26:
@britta-wstnr the default of `make_lcmv` is `weight_norm='unit-noise-gain'` and it is not set above, so it should be normalized.

We should look into this at the sprint.

britta-wstnr, 2019-04-22:
@lbailey25 does this problem still exist? If so, can you share the data so we can have a look?

CC @wmvanvliet @ckiefer0

larsoner, 2019-04-25:
@britta-wstnr pushed some fixes that probably fix this, I'll close but @lbailey25 please check and let us know if it's better (we'll reopen if it's not!)

closed 2019-04-25 by larsoner