mne-tools/mne-python #5267: adding Brain object to html report

dnacombo, 2018-06-07:
If I pass a brain object to report.add_figs_to_section, I get the following error:

```
Traceback (most recent call last):

  File "<ipython-input-186-7ad46ff9ab2f>", line 1, in <module>
    runfile('/lena13/home_users/users/chaumon/owncloud/Lab/Projects/CTPS/Datalink/MNE_analysis/code/plot_forward_sensitivity_maps.py', wdir='/lena13/home_users/users/chaumon/owncloud/Lab/Projects/CTPS/Datalink/MNE_analysis/code')

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/lena13/home_users/users/chaumon/owncloud/Lab/Projects/CTPS/Datalink/MNE_analysis/code/plot_forward_sensitivity_maps.py", line 75, in <module>
    report.add_figs_to_section(fig, captions=['test'], section='test')

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/mne/report.py", line 950, in add_figs_to_section
    comments=comments)

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/mne/report.py", line 905, in _add_figs_to_section
    img = _fig_to_img(fig, image_format, scale)

  File "/home/anaconda3/envs/mne/lib/python3.6/site-packages/mne/report.py", line 77, in _fig_to_img
    if fig.scene is not None:

AttributeError: 'Brain' object has no attribute 'scene'
```
A running example, modified from plot_forward_sensitivity_maps (last lines):

```

"""
================================================
Display sensitivity maps for EEG and MEG sensors
================================================

Sensitivity maps can be produced from forward operators that
indicate how well different sensor types will be able to detect
neural currents from different regions of the brain.

To get started with forward modeling see :ref:`tut_forward`.

"""
# Author: Eric Larson <larson.eric.d@gmail.com>
#
# License: BSD (3-clause)

import mne
from mne.datasets import sample
import matplotlib.pyplot as plt

print(__doc__)

data_path = sample.data_path()

raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
fwd_fname = data_path + '/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif'

subjects_dir = data_path + '/subjects'

# Read the forward solutions with surface orientation
fwd = mne.read_forward_solution(fwd_fname)
mne.convert_forward_solution(fwd, surf_ori=True, copy=False)
leadfield = fwd['sol']['data']
print("Leadfield size : %d x %d" % leadfield.shape)

###############################################################################
# Compute sensitivity maps

grad_map = mne.sensitivity_map(fwd, ch_type='grad', mode='fixed')
mag_map = mne.sensitivity_map(fwd, ch_type='mag', mode='fixed')
eeg_map = mne.sensitivity_map(fwd, ch_type='eeg', mode='fixed')

###############################################################################
# Show gain matrix a.k.a. leadfield matrix with sensitivity map

picks_meg = mne.pick_types(fwd['info'], meg=True, eeg=False)
picks_eeg = mne.pick_types(fwd['info'], meg=False, eeg=True)

fig, axes = plt.subplots(2, 1, figsize=(10, 8), sharex=True)
fig.suptitle('Lead field matrix (500 dipoles only)', fontsize=14)
for ax, picks, ch_type in zip(axes, [picks_meg, picks_eeg], ['meg', 'eeg']):
    im = ax.imshow(leadfield[picks, :500], origin='lower', aspect='auto',
                   cmap='RdBu_r')
    ax.set_title(ch_type.upper())
    ax.set_xlabel('sources')
    ax.set_ylabel('sensors')
    plt.colorbar(im, ax=ax, cmap='RdBu_r')
plt.show()

plt.figure()
plt.hist([grad_map.data.ravel(), mag_map.data.ravel(), eeg_map.data.ravel()],
         bins=20, label=['Gradiometers', 'Magnetometers', 'EEG'],
         color=['c', 'b', 'k'])
plt.legend()
plt.title('Normal orientation sensitivity')
plt.xlabel('sensitivity')
plt.ylabel('count')
plt.show()

from mne.report import Report
report = Report(title = 'Forward')

fig = grad_map.plot(time_label='Gradiometer sensitivity', subjects_dir=subjects_dir,
              clim=dict(lims=[0, 50, 100]))
report.add_figs_to_section(fig, captions=['test'], section='test')

```

larsoner, 2018-06-07:
As a workaround you can construct the `Brain` using a Mayavi figure handle, then pass that handle to `add_figs_to_section`. But we can add support for a `Brain` instance, too.

agramfort, 2018-08-04:
@dnacombo use:

```
brain = grad_map.plot(time_label='Gradiometer sensitivity', subjects_dir=subjects_dir,
              clim=dict(lims=[0, 50, 100]))
report.add_figs_to_section(brain._figures[0], captions=['test'], section='test')
```

closed 2018-08-04 by agramfort